ClockSelect/myevic #92: During LSL The Charging display its not available

afol2, 2016-10-20:
First of all, thank for your support, i think this is a astonishing effort you have in this Firmware, big fan of you by the way.

I have found that during LSL its "ON" on my EVIC VTC Mini, after you plugged the USB cable to charge the device, the screen that tell you Batt Volt, Temp and % of the battery its not shown. I'm guessing its because the device do not recognize it is light sleep mode, isn't there a way it could be done ?

ClockSelect, 2016-10-21:
Light or Deep Sleep do not make any difference.
It depends on the Screen Saver and/or Stealth Mode settings.
I fyou configure a Screen Saver, the box will enter sleep mode after the duration of the screen saver. If you activate Stealth Mode, it won't be shown either.

Also see #85 on this topic.

ClockSelect, 2016-10-21:
BTW: This does not prevent the box from charging. Cells will be charged normally wathever the display.

ClockSelect, 2016-10-21:
I modified things, so that behavior should be more intuitive.

ClockSelect, 2016-10-22:
This issue is more or less a duplicate of #85

closed 2016-10-23 by ClockSelect