ClockSelect/myevic #93: Locked Resistance is changing

dabugen, 2016-10-22:
Hi,

with the latest firmware from today, I am having the problem that my locked coil resistance is changing a lot after the device was in standby. For example: I am in TCR M1 mode and my resistance is locked to 0.6 ohm (with the lock sign), everything is working fine. Now my Presa 75 TC goes into standby mode and I power it up again 5 minutes later, now (not always, but so far in 80% of the cases), the resistance of the coil has changed, for example to 0.8 ohm or 0.4 ohm, but it never asked me the "new coil question" nor did I change anything - it just is changed after powering up from standby. I didn´t have this issue before with the MyEvic firmwares. Any ideas?

Thank you,

Hans:)

StepWi, 2016-10-22:
Similar problem, but only with Cuboid Mini, with all Firmwares from the day: Resistance changes to 0.00, when fired "No Atomizer". After some reflashing and resetting I had a resistance value until screen saver began. I have this problem in all modes.
Reverted to the Version from 17th Octobre and everything is fine.
Evic Mini and Cuboids are working fine.

dabugen, 2016-10-22:
Yes, that´s exactly what I mean too, exactly that happens to me too.

afol2, 2016-10-22:
Evic VTC mini here, its also happening on Power Mode, the coil resistance its fluctuating a lot, and sometimes it gives me atomizer low, right now it says 0.19, but after i press the fire button it says 0.22 dont know why

dabugen, 2016-10-22:
Yes, same here too, real time resistance in the "res" line is fluctuating like crazy, didn´t do that before.

ClockSelect, 2016-10-22:
The "New Coil" question is only asked when the resistance is *not* locked.
This is standard behavior.

As for variation in rez value, this issue is addressed in issue #89 and will be fixed soon.

dabugen, 2016-10-22:
The resistance IS locked(!), still, it just changes it, it does not even ask the "new coil" question, it just changes it completely to odd values (like 3 ohm or 0 ohm - and then also says "atomizer short") after it was in standby mode.

Also, the "Res" line (3rd line in TCR M1 mode) that measures resistance all the time, jumps totally odd, from 0 to 3 or even 6 ohm within just seconds, it´s completely off.

I´ve switched back to MyEvic version from 16. October now and everything is perfect there. Resistance stays correct and locked and the "Res" in the 3rd line just is smooth and correct too again. This is definitely a issue with your firmware, because when I re-flash todays (22. October) firmware, the problem can be recreated easily.

ClockSelect, 2016-10-22:
And now?

ClockSelect, 2016-10-22:
As I said, when the resistance is locked, the FW changes its value silently when you change atomizer (or when resistance reading is skewed - which was the case). The "New Coil" dialog can only be display when the resistance is not locked. It has always worked this way.

dabugen, 2016-10-22:
I´ve just flashed it and it seems to be fine again now! I will report back how it did go the next hours. Thanks a lot.

afol2, 2016-10-22:
@ClockSelect for me, in power mode its fixed with the update, thanks.!

dabugen, 2016-10-23:
It was all good for the last 7 hours, no problem anymore, thanks for the good work and please get something up so that we can donate:)

StepWi, 2016-10-23:
For me the problem is solved. Cuboid Mini works as expected. Thank you, the firmware is fantastic.

ClockSelect, 2016-10-23:
Seem we can close.

closed 2016-10-23 by ClockSelect