ClockSelect/myevic #95: Hi ,planned support for xcube ultra?

roma1141r1, 2016-10-24:
Mod: Xcube ultra.
Controller: nuvoton nuc220le3an. bootloader is not free but they are not required.
Interfaces: bluetooth, usb, mabe i2c screen...

roma1141r1, 2016-10-24:
He also has a balancer battery, rtc, vbat, and can be 220W power ..

roma1141r1, 2016-10-24:
And RGB led light... :) and vibro.... a am have nu-link programmer
I want to be Beta testers. I HAVE VERY LITTLE EXPERIENCE ida pro and hex rayse...
And c, c++ (stm32, cube bery little cmsis, win32)

roma1141r1, 2016-10-24:
I am also an electrical engineer (siemens s7) and have the opportunity to test the firmware via the programmer nu-link, although this is my everyday boksmod xcube ultra) but I have the original aprom in the case of return.

roma1141r1, 2016-10-24:
My mail is roma1141r1@gmail.com

ClockSelect, 2016-10-24:
Can you get the firware ouy of the MCU?

roma1141r1, 2016-10-25:
Yes, i am have firmware (only aprom), and all (ldrom, aprom, dataflash) wipe, and flash. This aprom https://yadi.sk/d/7fyPrUkgxYwrv

ClockSelect, 2016-10-25:
I'm afraid architectures and chipsets are too different for this box to be supported.
Best you could do is to to try to adapt the firmware to your chipset if you know how it's designed; this would require some reverse ingineering.

roma1141r1, 2016-10-26:
ok I will try to do it, do not know will or not. Can I ask you a question if I do not understand something?

ClockSelect, 2016-10-26:
sure.

closed 2016-10-26 by ClockSelect