ClockSelect/myevic #96: Smooth damping backlight

rocknrollman, 2016-10-25:
Hello! Thank you for the new releases, the firmware becomes steeper and steeper! I was wondering, can add beauty to smooth damping backlight like on dna plates? I`m have VTC and cuboid for tests.

TBXin, 2016-10-25:
There are no possibility to make fadein/fadeout like dna has.
You can change contrast from 0 to 100 on your device and you'll see that changes are minimal.
Also, even with a zero-contrast display is fairly bright.

ClockSelect, 2016-10-26:
The SSD1306 has a fade-out command (0x23) but does not seem to work. And the SSD1327 doesn't have it anyway.
A simple contrast fading gives an aesthetic result, even if not fading to black; I'll release it next time, you'll tell me.

ClockSelect, 2016-10-27:
Done

rocknrollman, 2016-10-27:
I like it, thanks ^^

closed 2016-10-28 by ClockSelect