ClockSelect/myevic #97: Long fire protection

hunadamka, 2016-10-25:
**Box Model:**
Joyetech eVic VTwo Mini HW v1.00 FW v5.04
**Coil setup:**
2.5ohm  Kanthal 0.15mm (9 loop on silica wick)

Hi! 
i'm using Kanger Protank 2 Mini
this is my best options:
Volt 3.5 (5 watt)

so i'm a cold vaper ...
and for me 10 sec Long fire protection not too match 
can u change to 15 or 20 sec? 

Thx!

falconettt, 2016-10-25:
I suggest to implement  "pocket protection" through some parameter other then TIME.  May be the dynamics of current or voltage or resistance or board temperature. You know better how to do it.

But let people have "max inhale time" as they want.

As for me, I want it to be something about 5 sec.
But I don't like to get that "long fire protection" screen  every time at the end of inhale.
And I obviously don't want mod to go off after that as it is now.

sorry for my english,

SEEKNDESTROY, 2016-10-25:
Why not let go of the fire button and then pressing it again real quick before the 10 sec. long fire protection is triggered? The 10 sec cutoff shouldn't be messed with in case a beginner doesn't know what they are doing.

hunadamka, 2016-10-26:
ok so you can't understand my coil setup
i don't like warm vape thats why need time for 
this setup = cold vape (MTL)

but yeah if you a cloud chaser you never understand this...

your first sentence is a  last solution 
i do this way almost 6 years : )

hunadamka, 2016-10-26:
and of course if somebody use cotton wick never use more than 10 sec 
if you bring in your pocket and don't turn off that kill your cotton easily!

ClockSelect, 2016-10-26:
I removed the "long fire protection" screen when the protection is configured below 10s.
The switch-off feature only triggers when the button has been continuously pressed 10s after the protection. Should not occur in normal usage.

hunadamka, 2016-10-26:
this is not help for me 
pls solve my problem too

ClockSelect, 2016-10-26:
Ok, let's try 15s.

hunadamka, 2016-10-26:
Cool!
You Are My Hero! : )

falconettt, 2016-10-26:
I'm terribly and extremely happy, thank you...

closed 2016-10-26 by ClockSelect