ClockSelect/myevic #100: Wismec RX 1/3

falconettt, 2016-10-26:
A situation may arise when you have only one battery. Is it possible to make box work in this situation? Now only 2 or 3. Unless of course you're gonna think this idea is interesting...

bluenazgul, 2016-10-26:
Only one Battery is not possible without Hardware modification

ClockSelect, 2016-10-26:
Does the box light up when shortcutting the second battery placeholder, and closing the backcap?

Filo301, 2016-10-27:
@ClockSelect Yes, it does light up!

ClockSelect, 2016-10-27:
@Filo301 I just have some code to *remove* do do that :)
What would do you use as a fake shortcutting battery cell?

Filo301, 2016-10-27:
@ClockSelect  I dont have rx200s, but it's interesting feature. For test i used short cable.

bluenazgul, 2016-10-27:
I think there is no need for two Batteries on the RX200S or a single Batterie on RX200S / RX2/3 as it need an "Hardware Modification" on both devices

ClockSelect, 2016-10-27:
@bluenazgul : no, it just needs a dummy battery that shortcuts the connector, just like the backcap of the RX/2 does when it's used in dual-cell setting.

Something like: [image]

The problem is not there.
The problem of thoses boxes in single-cell mode is that there's no boost dc-dc converter, so the voltage would be very limited and the bow would only vape in low powers like, maybe, 15~20W.

ClockSelect, 2016-10-27:
I uniformized the battery management code between the two boxes so now you can do what you want.
To reduce the number of cells, you have to short the 3rd battery connector to go dual-cell, and the 3rd and 2nd to go single-cell.

On RX2/3, the 3rd is the rearmost (the one in the backcap), the 2nd is the right one when looking into the battery compartiment.

On RX200S, the 3rd is the right one in the compartiment, and the 2nd is the rearmost.

Filo301, 2016-10-27:
One cell configuration does not work :( rx didn't read atomizer resistance.

Janishewskiiy, 2016-10-27:
It would be nice to provide for and possibility of parallel connection of batteries. This may require increasing the current limit.

Filo301, 2016-10-27:
@Janishewskiiy but probably impossible in rx200s/23/etc.

ClockSelect, 2016-10-27:
Serial or parallel changes nothing through a dc-dc converter.
P=UI ; halve U and double I, what do you get?

Filo301, 2016-10-27:
No balancer needed and in rx max 3v on output.

falconettt, 2016-10-27:
WOW!
It remains to find an elegant way to short the contacts and it's done. 
Do I understand correctly that to make box work at all without a battery in the passthrough mode is impossible?

ClockSelect, 2016-10-28:
@falconettt : yes; circuit must be closed if you want the current to run through.

ClockSelect, 2016-10-28:
@Filo301: did you retry with latest commit?

Filo301, 2016-10-28:
I retry but with no luck, still "no atommizer"

falconettt, 2016-10-28:
Yes, RX2/3 in dual-cell mode, right battery shortened.
Mod shows coil 0.00 and "no atomizer"

falconettt, 2016-10-30:
The miracle did not happen...

closed 2016-10-30 by falconettt