ClockSelect/myevic #99: RX200S logo

XqqsMe, 2016-10-26:
Hi,
Just updated custom firmware. Excellent job! I was trying to set up to show custom logo but it's not having it on screen even though it's selected. Other screen savers such as snow are working fine. Any help is appreciated. Also will the clock and date and other things lose their setting when changing batteries? Thanks

bluenazgul, 2016-10-26:
Logo's have to be 64*40 atm (may change soon) - since we dont have an RTC Clock Settings may loose on chanaging batteries

XqqsMe, 2016-10-26:
So the bmp files will not work, correct?

bluenazgul, 2016-10-26:
the have to be bmp (64*40 Pixel monochrom Bitmap)

XqqsMe, 2016-10-26:
Mine are and work fine with the Wismec latest update. I've uploaded to device but new evic software not showing it.

bluenazgul, 2016-10-26:
That is because the Wismec RX200S uses default 64*48 Pixel Logos and the current myevic release uses smaller Logos (only 64*40 Pixel)

This will change in future when @ClockSelect has the it done

ClockSelect, 2016-10-26:
Logo stuff has been developped and will be released soon.
For the clock thing: real-time information is lost when changing batteries if you're not fast enough (around 10s). But you can change batteries while plugged on the USB and time will be kept.

XqqsMe, 2016-10-26:
Ah. Gotcha now. Sorry for the confusion.

XqqsMe, 2016-10-26:
Good info on the battery issue. Keep up the good work!

ClockSelect, 2016-10-26:
48 lines Logos should work, now.

XqqsMe, 2016-10-26:
Thanks. I'll have to get new bin file correct?

ClockSelect, 2016-10-26:
correct.

afol2, 2016-10-26:
64x48 should work for cuboid/evic too ? i dont know id the dimension of the screen are the same in RX200s / RX2/3 and the cuboid/evic @ClockSelect

XqqsMe, 2016-10-26:
Updated my bin file. 64x48 bmp logo on RX200S working great! Thanks!

bluenazgul, 2016-10-26:
Now only the NFirmwareEditor needs to get Compatible with the supported Logo sizes.

ClockSelect, 2016-10-26:
@afol2 : they should work on all 14 supported boxes^^

afol2, 2016-10-26:
Cool..! @ClockSelect

closed 2016-10-26 by ClockSelect

XqqsMe, 2016-10-26:
Just noticed that wattage on the RX200S only goes to 91 and not 250.