ClockSelect/myevic #102: Presa TC 100

OlafKroll, 2016-10-27:
If I press shows the box 50 watts, however, only felt 20 watts arrive.
This happens every third or fourth time.
With the original firmware does not happen.

ClockSelect, 2016-10-28:
I hardly see how it can be; do you have another single-cell box where you can install the firmware and the same ato setup, to see if the issue is specific to the TC100 or not? I'd really need some more informations if I hope beeing able to investigate.

dabugen, 2016-10-28:
Hi, I´ve finally got my Presa TC 100 today and already flashed it, seems to work fine for the last 10 minutes at least, but I will report back, also if I get the same issue like the topic creator.... (so far not)

dabugen, 2016-10-28:
First 10 hours done, everything just as perfect as I could imagine with your firmware. Am in TCR M1 mode with custom settings (350) for Titan 0.34mm, works like a charm, absolutely happy!

ClockSelect, 2016-10-30:
@OlafKroll : What's your setup? Does it happen with every atomizer/coil?

OlafKroll, 2016-10-30:
I tested only on one Setup!
Tomorrow i will test a new config!

dabugen, 2016-10-30:
First 24 hours+ are done now with my new Presa 100 TC and it has been going all fine. Are you also using TCR M1 mode or just Power mode?

OlafKroll, 2016-10-30:
only Power

ClockSelect, 2016-10-31:
@OlafKroll : Does enabling/disabling or tweaking the preheat feature changes something?

OlafKroll, 2016-10-31:
I flashed yesterday the latest bios.
Since then, the device fires properly.
For me that is thus positively done.
Thank you for your attention.

ClockSelect, 2016-11-01:
What worries me is that I've done *nothing*.
But I had a report of this very same issue on a forum, user saying it happens only when he uses preheat feature.  So... let's keep looking.

OlafKroll, 2016-11-01:
I also had the Preheat function turned on as this happened

ClockSelect, 2016-11-01:
I suspect something in the buck<>boost transition of the dc-dc converter.
If you show the battery voltage next to the battery, and the at volts on the third line, you should be able to see when the atomizer voltage steps above the battery voltage (happens at powers ~35W+). If it's what I suspect, it should happen at this moment.
The moment of tranqition depends on several factors (inhalation strength, battery charge, etc) so the event may happen randomly when vaping between 30 and 40W. The preheat may just help crossing the line.
Could you check that?
Unfortunately I've no Presa by hand.

ClockSelect, 2016-11-02:
You may give a try to the last commit.
Couldn't test it on Presa unfortunately, so I can only hope it'll solve the problem.

OlafKroll, 2016-11-02:
Iam always above 40W.
My settings.... 65W (Preheat) and 50W Vape.
And the Akku was full.

ClockSelect, 2016-11-03:
So? Did you see an improvement?

Daveee10, 2016-11-03:
On presa 75w the issue is solved. No problem anymore. Waiting the answer for presa 100

OlafKroll, 2016-11-03:
No problem anymore...... to

closed 2016-11-04 by ClockSelect