ClockSelect/myevic #103: add more game?

damian140920, 2016-10-27:
maybe you add more simple games like snake ?

Filo301, 2016-10-27:
better arkanoid cause we use only two buttons.

ReaLx3m, 2016-10-27:
I vote Battlefield 4 ... and while youre at it add multiplayer too... I have few more suggestions, but first lets have this done, and before you know it we will all be selling our PS4's

damian140920, 2016-10-27:
I think that the connection of the USB gamepad is possible ..... so gta 5 on e cigarette is coming!

ClockSelect, 2016-10-28:
Well, I'm afraid it's not planned in a near future; I have some project in-between more related to vaping quality and I think I'll focus on it.

anderlobo, 2016-10-28:
You could add more screen saver

dabugen, 2016-10-29:
Please focus on vaping quality, if I want to play games, I´ll use my PC or smartphone! Come on guys, this is a e-cigarette!

fonix232, 2016-10-30:
I think even the Vapy Bird is unnecessary here. This is a freaking e-cigarette, not a handheld gaming platform. Do you want to vape, or do you want to game? If latter, go buy a used PS Vita or Nintendo 3DS...

ClockSelect, 2016-10-31:
I'd be prone to agree.
Flappy Bird, albeit a fun technical performance, was not a so good idea in the first place.
I keep it because it uses few ressources, but it may not last forever. APROM size is very limited.
Anyway, I do not plan writing games ATM. But remember you can fork the project if you wish :)

closed 2016-11-02 by ClockSelect

SEEKNDESTROY, 2016-11-02:
I wouldn't be sad if you remove Flappy Bird. I use my mod to vape and not to play games.