ClockSelect/myevic #104: Rx2/3 fast discharging

valador, 2016-10-27:
Hi, i flashed my rx2/3 myevic firmware  and i sii, my mod more fast discharge battery (almost 30%) compared with official firmware. And in official firmware bettery 100% in myevic 95%. This is normal? Same coil, same atomizer, nothing changed. 3 battery mode. Charging battery only in nitecore d4.
And add please editable menu for cutoff voltage for more autonomy. 
Many thanks

ClockSelect, 2016-10-28:
I see no reason why this would happen, really.
Di you select a battery model in the expert menu? This modifies the way battery percentage is displayed,
but nothing else. This may give a feeling of greater discharge because curves are less optimistic than the original generic battery curve.

I could lower the cut-off from 3.1 to 3.0 but there's almost nothing to gain in autonomy, except maybe if you vape around 10W. This would be surprising with an RX2/3.

valador, 2016-10-28:
I am select 25r in expert menu at once, battery very fast discharging, very.
Editable cutoff voltage this is great :) But, mod discharge to zero, i plug battery in charger nitekore d4, charger shows 1 - 3.2V, 2 - 3.2V, 3 - 3.18V, not 3.1...

dywersant, 2016-10-28:
You have account for voltage drop under load.  You will not discharge down to 3v, unless maybe you vape at 5W

valador, 2016-10-28:
i vape in 70W

ClockSelect, 2016-10-28:
Voltages reading of the RX2/3 are not really accurate.
If your charger gives you reliable voltages, you may use the BVO feature in the Expert menu to compensate for the error offset of the box.

valador, 2016-10-28:
i see interesting thing, if correcting voltage battery throw BO in expert mode, % battery very high now, almost +- 0.03v changing give +- 10% batary status.

ClockSelect, 2016-10-28:
It depends on where you are on the discharge curve. There's no much difference between 4.0 and 4.1, but a lot between 3.55 and 3.65, for exemple.

closed 2016-10-30 by ClockSelect