ClockSelect/myevic #105: Cuboid max 92 Watts

jram0421, 2016-10-27:
Hey, in power mode on the Cuboid, I'm only showing  a max of 92 watts, can you fix this?

LordNex, 2016-10-27:
Strange. I'm not having that problem. Are you sure it's not your resistance of the coils in your atty?

ClockSelect, 2016-10-28:
Your resistance must be 0.15 or 0.89 I guess?

It must be between 0.32 and 0.40 to have full access to the 200W

LordNex, 2016-10-28:
Exactly. Every device has its "sweet point" per say that due to ohm law will allow it full access. This still is in effect for regulated mods.

afol2, 2016-10-28:
Guess is the AMP limit..... 25 amp limit

ClockSelect, 2016-10-28:
Almost all boxes have a 9V/25A limit; only the triple-cells have a 50A limit.

jram0421, 2016-10-28:
Gotcha. My coil is I think the .5 ohm SSOC from Kanger. So that would explain it. Thanks!

closed 2016-10-29 by ClockSelect