ClockSelect/myevic #106: Suggestion. Cut off after X ml vaped

zalaross, 2016-10-28:
It would make perfect sense a feature like this. When using a steel tank and depending on the vaped ml feature, an option to set a cut off after X ml would be perfect. Can you add something like that?

ClockSelect, 2016-10-29:
I'm afraid all sugggestion like "Cut after X" wil have the same answer.
Duplicate of #64

zalaross, 2016-10-29:
Thanks for your reply. I haven't read all the issues and that's why I didn't know that. Maybe a flashing warning on the screen would be better than "cut after X".

Villista, 2016-11-03:
Good time to all! Really a very important addition to this project! Many people try to quit smoking using e-cigarettes. However, the electronic cigarette is not a factor of smoking one cigarette end! And vapor continues nipples mode! Reminder or lock would be a very great help for these people. And binding to the volume of the vapored liquid (ml) would be more correct!

ClockSelect, 2016-11-04:
People aiming to quit smoking only do not care about what custom firmware may exist for their box. People visiting this site and interessed on how they can upgrade their box with this software are *vapers*, not quitting smokers.
The features you're talking about would be a great addition to *stock firmware*. I agree with that. But let's be honest: if you're reading this, you're not someone that wants to quit vaping.

Villista, 2016-11-04:
OK. With this we can agree or argue. But the liquid flow calibration function (adjustment coefficient mL / kJ) should be added. If the counter is reset to 0 ml (long press Fire), reset the internal counter variable kJ. And after a certain amount of liquid flow enter it volume manually in the menu, it's easy to programmatically calculate the ratio ml / kJ!

falconettt, 2016-11-04:
@Villista : Use VVEL as it is. It's clear, simple and reliable. Clockselect doesn't wont to develop this feature at this time. May be later it would be possible to persuade him.

enricorov, 2016-11-04:
@Villista I'm sure you're not doing it on purpose, but you sound quite rude. No one gives a dime to @ClockSelect for developing this, so be respectful.

Villista, 2016-11-04:
Oh! I did not mean to offend the author or be disrespectful! Probably wrong translation from Russian made you think about it! I myself am a programmer and I understand that it is difficult to maintain the free projects! Once again, I apologize to the author!

enricorov, 2016-11-04:
No worries, just thought I'd point it out. Back on topic, I think that a mL limit per day would be quite unnecessary at this point. 

I think a mL/kJ computation feature would be more useful at the moment. I'll have a look at the code and try and see how it works.

zalaross, 2016-11-04:
A limit would indeed be unnecessary but a warning could be useful. This is just my thought from my point of view.

falconettt, 2016-11-04:
@zalaross : yes, programmable "warning screen" may be a good idea. 

You ask for 5ml warning.
Some one asked for 10a warning.
My ask, may be, about 40 C board temperature.

If you are really interested, think it over and open a issue.
I won't because "a lot of me" here...^^

zalaross, 2016-11-04:
Of course I'm interested but I don't think that the developer is interested. So i just keep enjoying this custom firmware as it is and feel thankful about all the hard work. If there was a way I would definitely donate some money for this work.

closed 2016-11-16 by ClockSelect