ClockSelect/myevic #108: Beware RX200S UCH working when it's off!!!!

badummtss, 2016-10-28:
Box Model:
RX200s
Coil setup:
Dual Fused Staggered Clapton 0.15 ohm (iJoy Combo RDTA).
Batterys: 
VTC5

UCH is turned off but when i playing with NFE or do other things with my RX over USB, it charged it a little.
Tomorrow it charged the Batteries from 87% to 90% percent.
I think the switch is not working right.
I reflashed the Original FW and then Back to the myEvic but no difference all the times.
Thanks

ClockSelect, 2016-10-28:
Well it seems that a few leaking current is unavoidable. But there's nothing here susceptible to put your batteries at risk.

MarkyAD, 2016-10-28:
Also, a battery cell's voltage will slowly rise as the cell comes to a resting state after use. A change of
a few hundredths of a volt is enough to make it look like its charging when its really not. Larger changes in resting state voltage (and voltage sag under load) can be indicative of a weak battery.

ClockSelect, 2016-10-29:
That's right; 87 to 90% is at most 0.02V more.
This can hardly be interpreted as a charge.

closed 2016-10-30 by ClockSelect