ClockSelect/myevic #109: Bypass in multi cell boxes

TheNikiz, 2016-10-29:
I noticed that on multi cell boxes the bypass voltage is higher the more cells there are. Seems like each battery adds 1-2 volts to the voltage. Is it supposed to be like that?

Picture: http://imgur.com/a/2rMG4

Mods: RX75, Cuboid and RX2/3. (3 batteries, 2 behaves the same as the cuboid.)

ClockSelect, 2016-10-29:
Yes.
Since 3 cells boxes do not have a dc-dc boost converter to up the voltage, it is limited by the number of batteries actually installed.

TheNikiz, 2016-10-30:
Alright, thanks! I was just confused about the weird voltages. I was expecting "emulated" 4.2V or 8.4V and 12.6 if in series.

Tehnik45, 2016-10-30:
RX200S, after flashing boxing from the start temperature of 38 degrees, 3-5 puffs and the temperature 53-56 degrees, any ideas why?

ClockSelect, 2016-10-30:
Batteries generate heat when you fire. Since there's 3 of them, that's some heat.
What's your setup?

closed 2016-11-02 by ClockSelect