ClockSelect/myevic #110: RX2/3 TC issue

TheNikiz, 2016-10-29:
I noticed that the temp control isn't quite working like it should on the RX2/3. I tested this with the RX75, VTC Mini and the RX2/3. For the atomizer I used the Goblin Mini V3 with a single SS316L fused clapton coil on it at 0.26 ohms. Each device had same settings on them and the exact same resistance locked. 40W, 220 Celsius and 0.26 ohms.  On the VTC Mini and the RX75, it worked perfectly. On the RX2/3, it hits temperature protection instantly and the power drops down to 0-10W.

NFE device monitor:
VTC Mini: http://imgur.com/a/dIAUs
RX75: http://imgur.com/a/5y0HN
RX2/3: http://imgur.com/a/ynUGH

Not sure about the Cuboid. It behaves the same as the RX2/3, but mine acts weird because the board is partly fried. (Dead voltage regulator, I charged the mod with USB and accidentally opened the battery door and that fried it..) The real time resistance reads 0.1 ohms higher than the other boxes.

Filo301, 2016-10-30:
I'll try it tomorrow.

ClockSelect, 2016-10-30:
Here is what I get on the RX2/3, dual cell setup, single SS316L 0.75 ohm.
Some oscillations, but nothing terrible; it's working fine.

[image]

It seems like you have some spurious temp readings (the two spikes) that cause the box to stop heating. Those two spikes can hardly be real temps; temp cannot rise that way when power is kept low.

Here is what I have on the VTC Dual, same config:

[image]

The Dual is much more smooth (and you feel it when vaping), it's by far the best box I have for TC so far. I really think it's due to the quality of the temperature measurement that seem a lot less noisy.

ClockSelect, 2016-10-30:
Here is another test on RX2/3 with another TC algorithm under developpement (3 cells setup, SS316L 0.75):

[image]

See the temp readings jumping all over the place from 180 to 250° for no reasons, although the power is kept steadily stable. This box seem to really have a problem with ADC noise.

TheNikiz, 2016-10-30:
Well, seems like I did something wrong on my RX2/3.. It seems to work now for some reason.
http://imgur.com/a/4GWlU

ClockSelect, 2016-10-30:
Better with pictures directly displayed^^

Maybe some bad contact with the 510?

ClockSelect, 2016-10-30:
BTW, I pushed the TC algorythms version - you may give it a try.

valador, 2016-10-30:
Sometimes TC drop to pewer an no gone back, set TC and again drop me to power, helps only if open\close battery cover. coil: core 2x04 SS 316L cover 0.1 nichrome.

ClockSelect, 2016-10-30:
The box grops back to power mode after testing the coil is it juges it not suitable for temp control. The test is a 30~40W fire during 0.2 seconds the first time it's fired. After that, the box wants to see an increase of at least 15% in NI and TI modes, and at least 5% on SS and TCR modes.

You can fake it by lowering the value of the locked resistance in the Coils Management menu.

dywersant, 2016-10-30:
Here are my results:
Rx200s
default:
[image]
boost:
[image]
Really nice feature, thanks :)

closed 2016-10-31 by ClockSelect