ClockSelect/myevic #111: Set a Amp limit in all modes

CustardFlan, 2016-10-30:
Hi, i'm wondering if you can implement a global limit of Amps, set a number, and, do not let the box reach X amount of Watts beyond this limit, or in temp control, or bypass mode, , also in Preheat function, no matter what resistance is in, i think this will be useful in case someone was playing with our mods, or, simply, take into account the Amp discharge limit of the battery, and do not get above this
Sorry for my bad english, not my native language

Also, thanks for the firmware, it's AWESOME!

**Box Model:**
RX 2/3, eVic VTC Mini, Cuboid, eVic AIO

**Coil setup:**
Mostly Kanthal and SS316

ClockSelect, 2016-10-30:
It's already the case.
Box will never go beyond 25A (50A for RX2/3 & EX200S) in any mode on the atomizer side.
The Amp limit on the battery side is also enforced and depend on the battery model (25A default).

CustardFlan, 2016-10-30:
Yes, but, what if one only want to use 10A and no more? I have 20A batteries, but, i don't want to overstress, i'm not only speaking about battery safety, it's about personal preferences and a little bit more customization

ClockSelect, 2016-10-30:
You have the option to limit the power, that's the only limit you truly need. It limits Amps and Volts by itself.

dabugen, 2016-10-30:
Ah, this might explain something... on the Presa TC 100, the wattage never goes to 100 Watts, most likely because of that amp limit. Can you remove that or make it changeable as suggested? I especially purchased 40A 26650 batteries for that (not the typical small 18650 ones) and would like to use the full potential. I´ve already done my own measurements and they are fine for pulsing at even 60A. So that explains while in TC mode and setting wattage to 100W, it never reaches beyond ~80 watt even though the coil could use it at that moment. The original Presa TC 100 firmware does not limit it like that. Please make it setable.

ClockSelect, 2016-10-31:
The limits of original firmware of TC100W are 25A, 9v, 100w. I did not change them. You should have access to the full power range with coils between 0.16 and 0.81 Ohms. In TC mode, the box only use the power it needs to warm-up the coil and keep it constant, so it may not systematicaly go up to 100W. And whatever the mode, if you use a coil below 0.16 or over 0.81, you'll be limited in power by Amps or Volts. I'm not supposed to behave differently than the stock firmware if you use the default TC algorithm. Just, the display may show more clearly the power limit enforcement.

closed 2016-11-04 by ClockSelect