ClockSelect/myevic #112: Exclamation Mark?

MoDuLah, 2016-10-30:
**Box Model: Cuboid**

**Coil setup: Dual Twisted Clapton Coils @0.29Ω**

I have starting to use  this myevic firmware the last few hours and i see an exlamation mark (!) next to the battery. But no matter what i do/change it doesn't go away.
Tried to look about it on the manual with no luck. 
Thank you in advance.

[image]

ClockSelect, 2016-10-30:
You're using th NFE Edition of the firmware.
This is a russian branch different from mine, and I do not work on it.
But AFAIK, the exclamation mark is her to inform you that the Light Sleep mode is On in the settings.
(Expert menu>LSL)

MoDuLah, 2016-10-30:
Oh ok. Thank you.
Sorry for the inconvenience.

closed 2016-10-30 by MoDuLah