ClockSelect/myevic #113: Realtime resistance off again

dabugen, 2016-10-31:
Box Model: Wismec Presa TC 100

Coil setup: Dual Coil Titan at 0.63 ohm total

Since the latest update from 22 hours ago, the realtime resistance is again wrong, it jumps up and down a lot but most often just shows a offset by +0.2 ohm. Flashing back to the previous firmware from about 26 hours ago, it is all fine again. Also verified the same happens on my EVIC VTC Mini, it does.

Link to firmwares:

https://www.sendspace.com/filegroup/5q3fpbPiRZ1OyTFj6rBG6Q

myevic (10).bin = the one with the wrong realtime resistance
myevic (9).bin = the working one where this does not happen.

Thank you!

ClockSelect, 2016-10-31:
Strange.
I'm using the very same binary (checked by downloading yours and comparing) on 4 different boxes (Mini, Dual, both RXs) and many different coils (but only Kanthal & SS) and I do not experiment this at all. At most the usual ±0.03Ω or so measure noise.
And inspecting the diff, absolutely nothing changed that could impact resistance measurements.
This issue is a great mystery to me.

dabugen, 2016-10-31:
Yes, very strange then. But I can re-create it again and again by flashing these 2 different versions. Should I make a video and show you?

ClockSelect, 2016-11-01:
I'm not sure it would help.
The 0.63Ω are given by an external ohmmeter?
What are the minimal and maximal numbers you see?
Maybe I don't have the issue just because I use very low TCR coils. Probing the coil heats it a little and Titanium or Nickel are much more sensitive to that. I must have some Nickel somewhere I'll give it a try.

dabugen, 2016-11-01:
Yes, by an ohmmeter at ~21°C.

Just flashed the "faulty" firmware again, numbers (at "coil") are 0.63 when not vaping and when vaping (220°C) it goes up to 1.08 ohm at the most. Just like always.

Realtime resistance (3rd line), however, always "jumps" around by +0.2 and back down again. Totally odd. Flashing the older firmware, it is all stable again.

dabugen, 2016-11-01:
Vapes are also really odd with that firmware, most likely because of the wrong measurements. Remember, I am in TCR M1 mode with titan at 0.63 ohm, 100 watt starting power. The vapes are weak for the first few seconds then they start to kick in FULL POWER, much to hot. Most likely related to that wrong resistance measurement as well. Switching back to the older firmware, everything is smooth and even. It jumps up to ~83 watt for the start, then settles around 30 watt for the rest of the vape, all good.

ClockSelect, 2016-11-02:
I made a change to increase accuracy of the real-time rez; it may help solve your problem. On my side, it achieves something like ±0.005Ω noise when stabilized.

dabugen, 2016-11-02:
Thank you! I´ve flashed it about 30 minutes ago and so far, it all looks good again. I will report back later on again:)

ClockSelect, 2016-11-04:
No news, good news.

closed 2016-11-04 by ClockSelect

dabugen, 2016-11-06:
All fine again, thanks!