ClockSelect/myevic #115: Device always switches to Power mode

phantasm, 2016-11-01:
1. When pressing Fire in TC mode, box immediately switch to Power mode
2. When I try to change mode to TC and press Fire, it also switches back to Power mode, I have to wait a few seconds in edit mode not pressing any buttons, so the device automatically applies TC mode. But when I vape, it switches to Power mode again.

Firmware: 7595147 (Temp Control: PID algorithm (experimental)) ClockSelect committed 2 days ago
Box Model: Wismec RX 2/3, using it with 2 batteries now
Coil setup:  micro SS 0.5 Ohm, prefer TC

afol2, 2016-11-01:
I have a friend with an RX2/3 that is having the same issue..

dywersant, 2016-11-01:
Fix your coils,  not related to firmware.

ClockSelect, 2016-11-01:
In SS mode, the firmware expects at least a 5% increase in resistance value after 0.2s of heating at 30~40W.
It's indeed impossible not to achieve this result with a correct micro-coil setup.

phantasm, 2016-11-01:
You mean that I have a short circuit in my coil?

phantasm, 2016-11-01:
Still, why does it immediately switch from TC to Power mode while resistance meter shows 0.51 so it's OK ?

SEEKNDESTROY, 2016-11-01:
Try a spaced coil. I'm running a single 24 gauge 316L SS, 11 wraps, 3mm inner diameter... Ohm's out at about 0.55 ohms.

afol2, 2016-11-01:
I could also say that, check if the screws are tight..

afol2, 2016-11-01:
@phantasm ok, reading what @ClockSelect says.. I talk to my friend that were having the same issue, and his RX2/3 in the 3th Line he put the Res "0.000" it was oscillating a lot, and it was not the same as the locked ohm... he tight the screws and reinstall everything again, and the Res match the locked ohm after that, and problem solved...

phantasm, 2016-11-01:
I'll make a new coil and write here about the results

Daveee10, 2016-11-01:
IMO the problem is complicated... I think the 24GA (0,5mm) SS too thick and not change the 5% resistance in 0,2s  what needed for check...

RenegadeTitan, 2016-11-01:
@ClockSelect i wrote a patch to fix this in original firmware. i am now experiencing the same problem with yours. the problem is the 30-40 watt range the coil check is performed at. it is an arbitrary number picked for the original 1 battery mods that only fire at 60-80 watts so doesnt have much of a negative affect on those. when someone uses any coil that requires more than 40 watts to heat up (a huge percentage of multi battery mod users that fire at up to 200-250w) this 30-40 watt range negatively affects every time coil is fired from room temperature by either slowing down warm up or in a large percentage of cases causing false positive non-temp coil reading if warm up is slow enough. this 30-40w range adds virtually no protection to any aditional user error scenarios compared to a test of ohm change performed at the user defined wattage protects from. please, please, remove the 30-40 watt range user defined wattage change that occurs durring the temperature wire coil check :) the wire check confirms faster if performed at user defined wattage.
p.s. i have been eagerly awaiting this rx2/3 firmware and absolutely love the functionality. i look forward to being to use it.

RenegadeTitan, 2016-11-01:
@phantasm i believe clockselect is correct about your coil. it may heat up a hair slower with 30-40w range but should have little enough mass to clear coil check. remove battery door, reattach. set to tc ss mode, when at room temp after 10m of sitting remove and reattachattomizer. note ohm reading. in coil settings set this noted number as new ohm setting. if this doesnt work check that negative pin and allan screws are tight.

phantasm, 2016-11-01:
I've made a new coil that has low resistance (0.28) and don't rise too fast when heating up. It seems that now TC is working fine, but I have to rise temp to 200C, which is a bit high for SS, I suppose. 
small video: https://www.youtube.com/watch?v=SNbx8Bye4os

I'll wait overnight to see if my coil will be OK.

phantasm, 2016-11-01:
@ClockSelect maybe it will be better to add some notification like "check your coil" or "unstable coil reading", or something like that, when the device needs to fallback to power mode.

Also, maybe it should set output power to something low (5w?) to prevent coil overheat, sparks or burning?

SEEKNDESTROY, 2016-11-01:
@phantasm I have mine set to 430f (221c) at 35 Watts. As I mentioned "I'm running a single 24 gauge 316L SS, 11 wraps, 3mm inner diameter... Ohm's out at about 0.55 ohms." Temp mode is working just fine.

RenegadeTitan, 2016-11-01:
2 twisted 24awg wires for core claptoned with 30awg. 5 wraps. dual coils. 0.09ohm. in tc with full airflow regulates at 65-75 watts. trying to reach temp with only 40w is futile lol

ClockSelect, 2016-11-02:
@RenegadeTitan: there is no longer any added warming time once the test is passed; it's just used once the first time it's fired. After that, on next fires, it goes straight to your power setting as long as you do not change the atomizer nor reset the box.

I agree with your analyzis; this issue of rez check with heavy coils has already be spotted. I have to find a way to prevent overtemp on light coils when the user have it's max power setting at max (remember that it's the default on stock firmware; also, one may forget to lower it when changing coil). I cannot take te risk of probing a microcoil @ 250W.

But let's do the maths.
In your case: there's really two tests done by the firmware; it wants an increase of 5%, but also a minimal increase of at least 10mΩ; since you're at a very low ohming, it's the last that takes precedence.
It means an increase of 92.6°C in temperature with a TCR of 120 (let's say 93°C since probe resolution is 1°C).
With a heat capacity of 480J/Kg/K @20°C for SS, your coils need 0.816J/°C; That's 75.9J total (let's says 76J).
In 2 seconds, it's 38W.
And we're giving 40W since your max setting is higher (the firmware fires at 30W if power setting is lower than 30W, at 40W if it's higher than 40W, and at the given setting if it's in between).
So it should barely pass the test (on the paper).
Considering a real-life situation, with some heat dissipation and readings inaccuracy (the RX2/3 is **bad** at reading resistance values), we're indeed very near the limits with your setup and it's enough to fail. Your experience is confirmed by science^^

But could we test at higher wattages? Remember the purpose of the test: it's here to protect you from bad things the day you forget to change the mode accordingly when you swap your ato. And you know it happens, so this test is needed even for experienced users.

So now say I swap to a regular .5Ω Kanthal coil with, say, a 5mJ/K heat capacity (typical factory coil). Testing it during 2 seconds at 30W is already enough to skyrocket it to 600°C if dry; things happen before that to tamper the heating due to the presence of liquid with some heat capacity and latent heat of vaporization, and barely saves the coil if you're lucky. But it's enough to feel something bad is happening if you're inhaling.
So we're stuck.

The best I'm thinking of ATM (tell me if you have some ideas) is to let you tell the box not to test the atomizer once you've screwed it in, like for example a manual check feature in coils management menu.

ClockSelect, 2016-11-02:
Found something while reading the code looking for ideas:

- Screw in your ato; don't fire.
- Lock your resistance (via editing the second line, or in coil management menu)

The test should be bypassed.
I can't really test it since I only have light weight build that pass the test instantly.

ClockSelect, 2016-11-02:
Well, I've added a bypass feature in the coil management menu.
More infos in the readme.

ClockSelect, 2016-11-02:
@RenegadeTitan : on heavy coils like yours, you may give a try to the PID algorithm.
(Menus>Vaping>Algo)

phantasm, 2016-11-02:
@ClockSelect I've already realised that my coils were not too good :) But thanks for the reply anyway!

RenegadeTitan, 2016-11-02:
@ClockSelect thank you so much for your quick response. i will respond with a thought out reply after i wake up a bit more. i just wanted to jump on to let u know quickly that there seems to be an error with new myevic.bin. nfe and wismec updaters both give errors.

phantasm, 2016-11-02:
@RenegadeTitan what error have you got? when I tried to upload new firmware today, wismec updater didn't recognize the device, and both NFE and VTCFont updaters fails to recognize it either. I had to reset the device a few times, play with USB ports, batteries and such, before it finally starts to detect. only then I was able to upload new firmware.

RenegadeTitan, 2016-11-02:
@phantasm are you sure you recieved a update successful message? error message from nfe was opened firmware is not suitable for the connected device. old myevic.bin works fine so that is only thing changing so most likely cause.

phantasm, 2016-11-02:
@RenegadeTitan yes, I got the new "Check" option in coil management menu and everything works fine. if you're out of other options, try to flash stock firmware and then upload myevic.bin again using wismec updater.

RenegadeTitan, 2016-11-02:
@ClockSelect apologies false alarm. should have checked file size. right clicking on file and save as in chrome results in ~45k file automatically named correctly (yesterday same procedure for .bin worked correctly) clicking on file and then view raw results in 91k file correctly downloading. this is what happens when i do things before ive had coffee lol

@phantask thank you for tips and letting me know the problem was me lol

RenegadeTitan, 2016-11-02:
@ClockSelect i havent had a chance to mess around with PID settings yet but am eager to. this is my favorite feature of this firmware (among many great features).

as for safety features the reason i feel 30-40w range can be removed without noticeably decreasing safety is that it requires 3 mistakes at once to come into play and still doesnt protect in this case(incorrect mode,wattage too high,dry cotton). if 1 or 2 mistakes are made ([dry cotton], [dry cotton and too high wattage], [too high wattage] not incorrect mode) there is not any protection. i believe wire check is useful for alerting user of incorrect mode but still doesnt provide protection. i have tested tiny coils at 250w wet for full 2 seconds and its loud crackling is a fair warning :) non temperature wire on dry cotton burns no matter what you fire at or what protection there is. to summarize: you cant offer any protection to mistakes in vw mode so why negatively affect vaping in temp mode to only try to protect same mistakes in vw mode that are also _mistakenly_ used in temp mode.

with that said i can understand the desire to protect 0.1% of error cases even if you cant offer any protection to 99.9% of cases (wrong mode/wire, too high of a wattage setting, wet enough coil that cotton doesnt combust but dry enough that 40w limit is enough to stop wire from melting where it otherwise would).

i think the setting to remove check would be adequate. it does not appear to be 100% effective tho. my device still cycles through 40w for almost a full second with coil check set to no. 

u stated that:

"there is no longer any added warming time once the test is passed; it's just used once the first time it's fired. After that, on next fires, it goes straight to your power setting as long as you do not change the atomizer nor reset the box"

i experience a different behavior. when my device sits for a few minutes (goes to sleep and coil cools down) when i fire it, the wattage displayed is 40w for about 1-2 second then set wattage. even with new setting, coil check no, it is still doing this every time. could firmware be avoiding mode change (to VW mode) result after first confirmation but still applying 30-40w range until ohm change is reached every firing?

if setting coil check no could entirely bypass coil check code including 30-40w code as opposed to automatically confirming passing of the test (as i suspect it may doing) i think it will greatly improve this speed.

imo how fast a device fires at set wattage from time button is pushed is very important for vape experience. 50ms is ideal. 200ms is on slow but acceptable side.

phantasm, 2016-11-02:
@RenegadeTitan even an experienced vaper can meet 1 or 2 criteria at once, ecpecially after a hard day or hangover :) a newbie can inflict damage to his setup more easily, so the safer firmware is, the better.

ClockSelect, 2016-11-02:
@RenegadeTitan : You're right, I forgot the case when the box wakes up from sleep mode. It resets the check too since it cannot know if you changed the atomizer when it was sleeping. I agree it partialy deafeats the purpose.
Well maybe I'll make the setting permanent, after all.

Concerning the warm-up phase at 40W max, it still exists, but lasts only 2ms, so should be not sensible at all. If you experiment a 2 seconds delay, it means the coil has been tested. If after that you go and see in the menu to find the setting on "No", it's normal since the test has been passed. And the delay should no more exist 'til the box sleeps again.

RenegadeTitan, 2016-11-02:
@ClockSelect as u know im ok with removing 30-40w range entirely for everyone as long as you keep actual coil check(at user defined wattage).  but i dont think thats necessary. if you use the setting check no, maybe just default to yes on firmware install but to retain no/yes in all other cases. then disable all cases changing it to no or yes except user input in settings. this should make the default that coil test is performed every firing (as it basically already does and as it does in original formware) but make setting check no permanent so that you can opt out of coil check for good. defaulting to yes on reset could get a little tedious every battery change.

infecti0N-113, 2016-11-02:
IMHO, may be is better able to do to "check" gave the same message as in the replacement of the atomizer:
**"TC Error! Check coil! Press left to "force TC mode", press right to "Power mode"**
or something similar) 

Sorry for my English =)

ClockSelect, 2016-11-03:
@infecti0N-113 : It's just another way to say "No" in the menu. It does not solve the sleep/reset problem.
@RenegadeTitan : Yes, screen saver is a good workaround^^ I think I'll mix all those idea to do something smart. Goal is to be as fluent as possible for user.

ClockSelect, 2016-11-03:
Test this.
It's a compromize that do not complicate life while still reasonably risky.

RenegadeTitan, 2016-11-03:
seems like maybe it has a shorter test requiring smaller ohm change? but it is still firing at 40w every time exiting sleep. it does seem to only fire at 40w for half the time before going to set wattage but imo thats still a half second too long. also it switched to vw mode on about 6th fire. this is first time it did that in a couple days. and i didnt wait for actual room temp which would cause even more likely false positives. sorry i couldnt give better review, i wanted to. ill load last version and set my screen saver back to 30m now :) i understand you want a user friendly way to accomplish this and i appreciate you working on it. and of course i have a lot of respect for the work you have put into this project. personally what i would like to see and i believe what many others would enjoy would be if mod would fire instantly at user defined wattage without delay every time fired. as for compromise, i understand if first firing on new coil has to be wire test. i am just concerned about false positives switching to vw mode on first fire. i liked the setting to skip the first coil check. and with screen saver basically always on device is functioning perfect imo. to avoid necessity for screensaver what about after each wake up a fast check of current ohm compared to set base ohm, to see if it is a new coil, instead of resetting the need for another 30-40w wire test? (if this could be done very quickly, which since its constantly checking ohm any way im guessing it can?)

ClockSelect, 2016-11-03:
Those are complicated thing.
I put back the Check option in a permanent version. Simpler.
And kept the new test, also. It's not shorter or anything, it just adjusts the power at wich the coil is tested to some proportion of the max power of the box. Assuming that users of more powerfull boxes uses bigger stuff, or at least know what they do.

phantasm, 2016-11-03:
@ClockSelect I'm using very light builds, like 0.1-0.5 Ohms or even this - https://www.fasttech.com/products/4813700 on my RX2/3 (wich has max power of 250W on 3 batteries). will the new test have issues with it?

RenegadeTitan, 2016-11-03:
@phantasm try it. report if you have any issues.

RenegadeTitan, 2016-11-03:
@ClockSelect i believe this higher range for test will reduce the issue of longer heat times for high wattage coils. however there may be a glitch in code as i am experiencing what seems like an unintentional operation. it appears that the limiting range (was 30-40w now is higher) is no longer reverting back to user defined wattage once wire is confirmed. i am experiencing around 40w firing, followed by ~80w firing until temp regulation kicks in. never reaching my set 120w. also temperature regulation maintains at 15 degrees f OVER what it is set at. tested in algo pid and off, same over temp behavior also check no setting never changes to no. appears to not actually fully clear wire test function and may be getting stuck somewhere in that function. performing wire test every fire regardless of if going to sleep or not since last fire.

RenegadeTitan, 2016-11-03:
@ClockSelect one behavior im noticing is that with setting check off is that when device reaches and maintains a temperature ~35 degrees f **above** set temperature. sometimes when firing again right after it reaches and maintains correct temperature. i need to test more as i so far cannot tell what conditions are triggering correct or above correct regulation. there appear to be these 2 separate and consistent behaviors, one being correct.

ClockSelect, 2016-11-04:
@phantasm : you should not have more issues than before^^

ClockSelect, 2016-11-04:
@RenegadeTitan : That would be another issue, if any.
Seems we're more or less done done with this check problem.

closed 2016-11-04 by ClockSelect