ClockSelect/myevic #116: Board Temp and Coil Temp

afol2, 2016-11-01:
Hi, I was wondering if there is a way, i can put the Board Temp in C, meanwhile the Coil temp in F. 
Maybe it could be round robing as i think it was in the original firmware.

-->

**Box Model: Evic VTC mini Firmware  7595147

**Coil setup: White Bone MINI RTA @ 0.22 ohm

ClockSelect, 2016-11-02:
The temp unit is selected for all temps display at once - it was also the case in the original firmware. You can't display vape temp in °F and board temp in °C. And why would you do that? It would be confusing.

The round robing has been replaced by a temp unit selection in the interface menu because it's far more handy for 99.99% of people that don't care about the other unit system. And it avoids common cotton-burning mistakes. You select your unit system  once and for all and I think most of the world is happy with that.

afol2, 2016-11-02:
Is just that... Where I live, we measure weather Temp in C ° but the coil I learn to measure in F and when i get the board temp in F °... I don't understand it at all. and if i put it in C °, i dont know if the coil its to hot or not.

ClockSelect, 2016-11-02:
So set the box in °F. Why do you care about the board temp, after all?
What's more, the board temp reading is completely unreliable.

closed 2016-11-04 by ClockSelect