ClockSelect/myevic #117: What to buy?

falconettt, 2016-11-01:
Old Evic mini with FW myevic is great but I need second one as spare.
What one battery mod is most accurate  and best suits your project? 
(My new RX2/3 found a new owner because it heats up when charging and vaping)

SEEKNDESTROY, 2016-11-01:
**RC200S is my personal favorite but then I don't charge and vape. I charge 3 sets of batteries in external chargers. Plus, Wismec does NOT recommend charging via USB port anyway. I think you should've kept the RX2/3 and invested in another set of batteries and an external charger like the [Luc 4](https://www.amazon.com/s/ref=nb_sb_noss?url=search-alias%3Daps&field-keywords=luc+4+charger&rh=i%3Aaps%2Ck%3Aluc+4+charger) IMHO.**

falconettt, 2016-11-01:
@SEEKNDESTROY  thanks for the advice, but I vape at 20w, and prefer to charge via USB.
One battery box and a couple of spare batteries IMO is all I need.
But what one battery box is most accurate and best suits myevic firmware?

ClockSelect, 2016-11-02:
The best for the project would be one of each of the 14.
Really, buy what pleases you more.
VTwo is a very nice box; you may also consider the Dual, for almost the same price. I like it because it's flawless in temp control. Very accurate and consistent readings giving a very stable vape. I guess VTwo must be almost the same, but did not really test. It depends on when in the design process they improved their components.
I don't know about all the other boxes.

lojpek, 2016-11-02:
I love my presa 75W TC for small size and button that I can press by hand (don't need to search my fingers for button). Only one con that you must spray it with color, because wismec have very poor paint.

falconettt, 2016-11-02:
Yes, presa 75W TC looks good on the picture, but black color makes me sad and poor paint is not to deal with.

eVic VTC Dual 
is more accurate in temp control (I'm happy without TC), but has a crack on the upper plane, which may cause the ingress of liquid inside.

So the eVic VTC Mini seems to be ideal.(25$)

But what about eVic VTwo Mini? Does it have any real advantages exept clock accuracy?

ClockSelect, 2016-11-02:
On the paper, a VTwo Mini is a VTC Mini with a clock, period.
After that, since it's more recent, they may have tuned the electronics in some ways. For example, the Dual is better than the VTC for resistance reading. There must have been some tuning in between. I can't tell about the VTwo since I don't have one by hand to test. I should see a TC monitoring report with PID algorithm to compare.

ClockSelect, 2016-11-02:
eVic VTC Mini, SS316L 0.75Ω @ 180°C, 75W:
[image]

eVic VTC Dual, same setup & config:
[image]

Obviously, they changed something.
I'd like to see the same for the VTwo.

falconettt, 2016-11-02:
So, "help wanted"
could anybody be so kind to give us
evic VTwo TC monitoring report with PID algorithm to compare?

enricorov, 2016-11-02:
I can help, but I'll need some explaination on what to test, and how to provide graphs.

falconettt, 2016-11-02:
@enricorov : you need to know how to manage with NFirmwareEditor but it's a bit difficult to explain in one sentence...

enricorov, 2016-11-02:
[image]

Don't know how to export graphs on NFE, so here's a screenshot.

PID algorithm, VTwo mini, TI build 0.15Ω @ 220°C, 60W.

falconettt, 2016-11-02:
@enricorov , great thanks, may be thats enough for @ClockSelect to to make the right decision...

ClockSelect, 2016-11-02:
wow... what are your PID parameters?
should be 600/850/0 by default.

enricorov, 2016-11-02:
My values are the default ones, that is 600/850/0.

ClockSelect, 2016-11-02:
So maybe it's just a false reading of temp. It happens.

Anyway, VTwo seems to have the same accuracy as the Dual.
Better than the VTC anyway.

ClockSelect, 2016-11-02:
@falconettt : it's your decision, not mine. I'm just collecting data to help^^

falconettt, 2016-11-02:
OK^^
anyway, thanks everybody!

enricorov, 2016-11-02:
[image]

Doing some more testing, the graph looks more like this.

enricorov, 2016-11-02:
It's up to you to choose a box, the VTwo is really nice, but I don't have any other for comparison.

afol2, 2016-11-02:
Ok, im trying to do that on my Cuboid, but the NFirmwareEditor its saying "No Compatible USB Devices conected" Mod is in COM and NFE is On

ClockSelect, 2016-11-02:
@enricorov VTwo looks like a good TC vaping mod.

ClockSelect, 2016-11-02:
@afol2: Use the usual procedure. Close everything, reset everything.

falconettt, 2016-11-02:
@afol2: if you are on win7 you need a driver, try to find it/

ClockSelect, 2016-11-02:
There is a driver directory in the repo.

afol2, 2016-11-02:
@falconettt yes window 7

falconettt, 2016-11-02:
@afol2: take it...
https://yadi.sk/d/R3-2wKchxzB8q

afol2, 2016-11-02:
@falconettt thanks.!

falconettt, 2016-11-02:
@afol2:
use here:
[image]

falconettt, 2016-11-02:
So, let it be eVic VTwo Mini...

closed 2016-11-02 by falconettt