ClockSelect/myevic #118: VTwo mini, unable to connect to PC

enricorov, 2016-11-01:
**Box Model:**

**Coil setup:**

Hi everyody, thanks for the great firmware.

I just flashed it, using the official tool, and everything seemed to be working fine. I tested NFirmwareEditor's Device Monitor, which correctly detected my box. 

Then I tweaked a few settings, and, in the 'Expert' menu, turned off USB charging. I don't know if that caused the issue, but even after turning it back on, I can't seem to be able to connect to my PC, with NFE or with the official tool.

Is there anything I can do?

Thanks in advance.

My box is an Evic VTwo mini.

The 'advanced' menu looks like this:

USB: HID
DBG: OFF
X32: ON
LSL: OFF
NFE: OFF
SHR: 115
VCH: ON
BAT: VT5 (changed this one too)

bluenazgul, 2016-11-01:
Set USB to COM and it will work again

enricorov, 2016-11-01:
It worked, thank you very much. I hope it wasn't me changing the setting accidentally.

StepWi, 2016-11-01:
And set NFE to on to make ist visible to NFirmwareEditor.

enricorov, 2016-11-01:
According to the documentation, that's for advanced tweaking. It's visible to Device Monitor in NFE even with that option off.

StepWi, 2016-11-01:
You need NFE set to ON for firmware update and use of the advanced tweaking. USB should be set to HID for that or COM if you want life view. AFAIK COM includes the HID setting.

enricorov, 2016-11-01:
I get it. Thanks for the info.

closed 2016-11-01 by enricorov

ClockSelect, 2016-11-02:
USB Charging has (should have, at least) no incidence on usage of USB softwares.
NFE ON is only needed to use NFE advanced features like Device Monitor, but may cause issues with some firmware updaters. The official firmware updater from your manufacturer may require NFE OFF. Usual features of NFE (font editing, firmware update, logo upload etc) are available in any case.
USB COM is needed by the NFE Device Monitor, but it should be able to enforce it if needed.
USB HID is always active, whatever the mode.