ClockSelect/myevic #119: my rx2/3 cant connect to NFE device configuration or device monitor

RenegadeTitan, 2016-11-01:
i saw that you posted a live readout of an rx2/3 firing so i assume im missing something. my rx2/3 is recognized while expert setting usb is set to pid. NFE says i need to enable vcom and this changes usb setting from pid to com. at this point NFE no longer recognizes my device. i saw u had links to what looked like mac virtual com drivers. do i need a windows virtual com driver for my pc or something? (running windows 7) what am i missing?

Daveee10, 2016-11-01:
Try to turn Off the UCH...

StepWi, 2016-11-01:
And make shure NFE is ON in Expert Menu.

RenegadeTitan, 2016-11-01:
thank you for your responses. i already tried both of those and just tried it again to no avail. im using the myevic.bin directly from this github. does that firmware support it?

RenegadeTitan, 2016-11-01:
i also tried clearing dataflash and reinstalling.

TBXin, 2016-11-01:
Hello,

Your are missing VCOM driver for Windows. You can get it in this rep ;)

ClockSelect, 2016-11-02:
Thanks @TBXin

closed 2016-11-02 by ClockSelect