ClockSelect/myevic #120: Voltage shown in Bypass Mode

cblag13, 2016-11-01:
I'm using an **Evic VTC Mini**, dual twisted Clapton coils, **0.34 ohms**.
Don't know if it this an issue or I didn't fully understand how it works, but...
In **Bypass Mode**, the first line voltage is **3.06V**, even though the battery voltage is **3.65V**. Also, the power, displayed on the top of the screen, is **27.5W** (instead of **39.1W**, how it should be at 3.65V). 
So...why?

ClockSelect, 2016-11-02:
If your battery shows 3.65V, you can't send 3.65V to the atomizer. You have to take into account the internal resistance of the battery, and the efficiency of the circuits. The displayed voltage/power is a predictive value; maybe not perfectly accurate, but far more than the raw values derived from sole battery voltage.
When you actually fire, what does it says?

cblag13, 2016-11-02:
For example, now (in Bypass Mode) the battery voltage is **4.17V**, the voltage shown in the first line is **3.74V**, the top screen wattage is **41.1W**and when I press the fire button, the wattage shown is between **39.2 - 39.6W**. The wattage at **4.17V** should be **51.2W** (hypothetical, leaving aside the internal resistance). It's kind of a big difference

ClockSelect, 2016-11-02:
*"leaving aside the internal resistance"*...

The **41.1W** is a fairly good approximation of the given **39.6W**.
Would you prefer that I display **51.2W** and give you **39.6W** like the stock firmware does?

Bypass mode is a WIGIWYG system (What You Get Is What You Get).
You cannot leave the internal resistance aside if you want to estimate what you'll get.

cblag13, 2016-11-02:
Ok, that sorts things out. Thank you

closed 2016-11-02 by ClockSelect