ClockSelect/myevic #122: Set clock time still jumping (unpredictably) when switching batteries

sidran, 2016-11-02:
I noticed we already had an issue with this and it was supposedly solved. I flashed myEvic firmware 2 days ago for the first time ever. Everything is brilliant except time which also seems to work fine till I switch batteries. Time offset goes from few seconds to more than 30 minutes to full clock (and date) reset. I tried switching batteries for testing sake, using half drained batteries and I could not perceive a pattern. Sometimes, just after inserting batteries, clock jumps around over 1-2 positions until it settles.

I am curious, so also please answer this if you know: Does clock save each tick into device's flash memory or it has some on-board, provisory, battery-like element which is used for limited time when main battery is unavailable?

Thank you so much for making this. Its awesome and with functioning clock it would be near perfect.

Box Model: Joyetech Evic VTC mini
HW version: 1.00
Firmware: latest myEvic
Battery: Samsung 30Q
LSL enabled, X32 disabled
TC mode used exclusively

Coil setup: 0.32 mm (gauge 28) SS316 @~0.56ohm on UberToot styled RTA.

sidran, 2016-11-03:
I just put freshly recharged battery.
When I started changing it was exactly 02:57:56
Switch lasted few seconds (10 at most)
Just after battery was inserted, for a brief time clock was at 00:48:xx (dont remember seconds)
Second or two later it settled at 03:02:xx 

Currently there is an offset of 320 seconds. This is pretty precise. When Evic clock hit 03:08:00 my computer's clock (to which eVic was synced through FWUpdater) was exactly 03:02:40

Hope it helps.

ClockSelect, 2016-11-03:
Thanks for report.
I'm aware of this issue and I think I know what it's linked to.
Will take some time to test and fix.

sidran, 2016-11-03:
I switched battery again and clock (freshly synced before change) now seems to be lagging exactly 5-6 seconds. Thats about how much battery change lasted.

sidran, 2016-11-04:
Next switch. Times were synced with FWUpdater as before.
After switching battery, which I intentionally did for ~10 seconds, time on my computer was: 05:01:56 while on the eVic it went forward to 05:20:00. Time is round because I am watching both clocks side by side at the same time.
As can be calculated, offset this time was: ~1084 seconds or 18 minutes and 4 seconds.

ClockSelect, 2016-11-16:
Do you still experiment jumps?

sidran, 2016-11-16:
I do notice them but stopped posting to avoid "spam" as you didn't ask for it.
Would you like me to keep doing it?
BTW, I didnt update firmware since last post. Should I do it?

ClockSelect, 2016-11-16:
Well, not each time youexperience it^^
It's just I've some hard time reproducing it on the VTC Mini.
I take off the battery, put it back, and... nothing. Just the usual few seconds lag.
It's not the first time I try since you opened the issue; It just seems to work fine on my side.

sidran, 2016-11-17:
Just updated firmware. Clocks were synced perfectly as usual (+/-1sec).
After changing battery, which took less than 6 seconds, offset came to 7 minutes and ~8 seconds (eVic ahead of time). 

I just discovered something while writing this.
**Have you tried genuinely swapping batteries? Like putting full after drained one.**
With this full battery I just inserted, I tried taking it out twice for 4-5 seconds and putting it back in. Both times offset was negligible, amounting to +/-5 seconds.

ClockSelect, 2016-11-18:
I think it depends on how long ago you changed the battery, so testing several times in a row does not reproduce the issue. Has to be tested.

ClockSelect, 2017-08-30:
I tried many things to make clock reliable on non-clock-enabled devices, and the best trick I found was to accept that the clock will be perfectly reliable for the time the battery will last. This is more a hardware than a software issue. Since any new device out there have the clock hardware, I won't work anymore on this subject. When I use my mini, I adjust the clock each time I need to do so, and I live with it.

closed 2017-08-30 by ClockSelect