ClockSelect/myevic #123: Egrip II and evic AIO LED setting

djmajo, 2016-11-02:
Hey There!
First: GREEEEEAT FIRMWARE! Thank you so much.

Now my problem: I can't get the Tank LED get to work. I have put the RGB Settings to 100% (each tested) but the light wont work. Is there a general "On/Off" switch in the Menu to get the Tank-light working? Couldn't find anything about it.... Only: SCREEN / MISCS / LED..... 

**Box Model:**
Egrip II and evic AIO

**Coil setup:**
0,6 Ohms CCell Coil in a Vapor Giant Go V2 at 50 Watts

Thank you!

Best regards from Germany
Mario

ClockSelect, 2016-11-03:
Yeah, I forgot something.
I have no eGrip II so I have to develop blindly.
test this: [prototype](https://www.dropbox.com/s/6878hw1w0ufei1t/myevic.bin?dl=1)
And tell me if it works.
Thank you.

djmajo, 2016-11-03:
Worked! For booth Devices. Egrip II and evic AIO.
Thanks a lot Bro!

Please contact me via Email. I have to tell u something good ;-)
djmajo@hotmail.de

closed 2016-11-03 by ClockSelect