ClockSelect/myevic #126: Battery % vs. Voltage

rsrmean, 2016-11-03:
Hi, 

First of all, thank you for this great mod. It's really fun to play with! 

I'm not sure what is causing it, but I think the battery % doesn't act like it should. It will stick around 60% from 3.8 to 3.7 Volt, then fall rapidly.

Do we have to calibrate the battery in some way? I didn't want to mess with the expert settings since I don't know the battery model inside. I'm using the Cuboid Mini. 

Best regards

StepWi, 2016-11-03:
It is very difficult or often nearly impossible to calculate the State of Charge (SOC) of a Lipo depending on the voltage of a cell. The cell in the cuboid mini isn't  changeable, and afaik of unkown type, so probably nobody knows the relation between cell voltage and SOC.
I have a similar "problem", it is showing 100% most of the time, even at 4.1V. I have set battery type to GEN(eric) in expert menu. I live with that limitation.
The only way to change behaviour is to calculate the integral of used mAh and compare it with the nominal capacity of the cell. But if cell is aging this calculation would be probably worse compared to the actual solution, or after discharge the nominal capacity has to be corrected to the calculated integral of used mAh.

rsrmean, 2016-11-04:
Thanks for the answer. Looking forward!

ClockSelect, 2016-11-04:
Done.

rsrmean, 2016-11-04:
Cool! I'll leave this open and report as soon as I can :)

ClockSelect, 2016-11-06:
No news, good news.

closed 2016-11-06 by ClockSelect