ClockSelect/myevic #127: base ohm reading

RenegadeTitan, 2016-11-03:
any chance the base ohm reading could be extended an extra digit to thousandths of ohm in coil settings instead of hundredths of ohm? i believe this would help a lot in calibrating displayed temperature readings and to a lesser extent slightly increase accuracy of temp control regulation and algo pid adjustments. my current coil is .096 ohms and when locked in at .09 it requires 350f setting to fire correctly or locked in at .10 requires 520f setting to fire correctly. quite a wide range for closest options to actual ohms.

ClockSelect, 2016-11-04:
Assuming TCR=120 and Rreal=0.096Ω,
In the first case (0.100Ω, 350°F) your target rez is 0.115Ω (185°C/365°F)
In second case (0.090Ω, 520°F) your target rez is 0.117Ω (214°C/418°F)
It's a bit different but let's assume you really want 200°C (392°F), we're around a target rez of 0.116Ω.
You want the rez to be 0.116Ω when T° shows 200°C
Say you lock your rez at 0.090Ω (better lower, avoid sub-zero temps).
You should be good by setting your TCR around 160.

Ok, it's a workaround.
But I must see what your change implies - rez are used everywhere in calculations and it's easy to forget something somewhere.

Problem in your case is that you use very low sub-ohm coils with low a TCR metal; combination of the two gives temp control a very high milliohm sensitivity (around 10~15°F/mΩ).

ClockSelect, 2016-11-04:
Before (10mΩ accuracy):
[image]

After (1mΩ accuracy):
[image]

Seems there's a little gain in temp stability, but that's not obvious with my setup (0.75Ω, I'm far from your needs).
Anyways, it works.

ClockSelect, 2016-11-04:
Give it a try.

RenegadeTitan, 2016-11-04:
@ClockSelect i went back and forth between firmwares about 15 times each relocking base resistance at .09 @ 520 degrees (your calculations say 420 but my experience is 520 is equivalent vape at .09) and .096 @ 420-430. seems to be nearly identical fluctuations as it regulates (on further thought seems it should not make a difference in functionality as long as you avoid higher ohm locked in than actual coil). possibly slight improvement but dont see how it would(again referring to functionality of regulation). one noticed behavior that i do not understand is that when on .01 accuracy firmware i lock resistance to .10 then immediately to .09 to remove any previous set base resistance. then flash to .001 accuracy firmware and .096 is automatically showing as locked (even if coil is warm and still above this) so it appears it is somehow pulling this from memory. maybe you used a different memory location/variable to function as base ohm reading? not an issue just found it interesting. 

as far as accuracy of displayed temperature this is greatly improved. allows much more room for calibrating lower ohm coils. mostly an aesthetic improvement but one that i enjoy a lot. i think this is a significant improvement. thank you once again :)

ClockSelect, 2016-11-04:
In fact, the "0.09" is stored as usual and the "6" elsewhere. Problem with dataflash is that if you wish to modify its data structure, you can only do so incrementaly, conserving content and meaning of previous locations, to allow compatibility between firmwares and versions of firmwares. Except if you accept a dataflash reset at some point.

Indeed, stability is not the point of this enhancement, but a little improvement may show up due to the PID algorithm. Since everything is know computed with mΩ accuracy, the proportional part of the PID feedback loop commands the power output by smaller steps. Before, it could only ask for ±15°F steps, and now ±1.5°F, so the resulting power/temp curves should be somewhat smoothed out.

Problem is that you're using an RX2/3 (or 200S can't remember but anyway) and those boxes have a very noisy reading of the resistance that cancels out the benefit of the measure accuracy in regard to the regulation stability.

RenegadeTitan, 2016-11-04:
small idea, maybe since most coils are under an ohm display on main screen .xxx accuracy unless over 1.00 ohm then display x.xx accuracy?  just an idea for if your looking for something to do :)

ClockSelect, 2016-11-04:
That's indeed an idea.

ClockSelect, 2016-11-04:
Here's what you should get if the resistance reading was less noisy (VTC Dual); you can compare with graphes above:
[image]

RenegadeTitan, 2016-11-04:
i thought live readings of coils were already .001 accuracy and compared to .01 accuracy of base resistance to get a .001 accuracy change. wouldnt this have same regulation?

RenegadeTitan, 2016-11-04:
comparing the 2 devices, what causes the noisy ohm reading, circuitry or 5/10?

ClockSelect, 2016-11-04:
Hm... Seems you're right. I'll cancel my previous comment.

Can't be 100% sure, but say 99% for circuitry. I have the same result with an RX that had his 510 replaced.

t2dYuri, 2016-11-04:
@ClockSelect thanks for the update. For now, i have locked 0.73Ω in the second row of main screen, and 0.739Ω in the coils management menu. I understand, that it's just visual differences, but may be it would better to observe the same value in both places?

RenegadeTitan, 2016-11-04:
:)

ClockSelect, 2016-11-04:
That was the point of 7 comments above^^

RenegadeTitan, 2016-11-04:
hey clockselect while your here mind if i ask you about nfe updater? i can pull info from device but cant upload changes to device. any ideas?

ClockSelect, 2016-11-04:
The configurator of the NFE is meant to be used with the NFE Edition of the firmware (an independant russian fork); your dataflash are not compatible. Maybe we'll do something for that in the future.

ClockSelect, 2016-11-04:
Display adapted.

RenegadeTitan, 2016-11-04:
ok thanks. i already removed that question and reposted as new issue. feel free to close it or delete it.

closed 2016-11-04 by ClockSelect