ClockSelect/myevic #131: Temp reading issue w/ latest update

enricorov, 2016-11-04:
I just updated to the latest build in the /bin folder here: the coil value is shown with one more significant digit, that is

OLD X.XX
NEW .XXX

That is great, but the temp read while vaping reaches values of 800/900 °C onscreen, with a 0.15 (cold) coil. I then reverted to the previous version of myevic and everything works flawlessly.

ClockSelect, 2016-11-04:
What's your setup?

ClockSelect, 2016-11-04:
Ok, I guess it's Titanium.
Would have gain some time if you told me.

ClockSelect, 2016-11-04:
Fixed.

enricorov, 2016-11-04:
I'm so sorry, I filed this while on the go, didn't think about it. 

It it indeed titanium, I'll try the new .bin and let you know.

enricorov, 2016-11-05:
Just updated, the bug seems fixed. I see you also removed the 'protection' warning when the coil reaches the set temperature target.

Thank you very much for the prompt reply.

ClockSelect, 2016-11-05:
Yes, the "Protection" indicator is just useless.
It just prevents your from seeing how far the target temperature has been overshot.
It "Protects" you from seeing that the default TC algorithm is bad.

closed 2016-11-05 by ClockSelect

enricorov, 2016-11-05:
I see, and think the same way. Thank you for all the effort you're putting into this firmware.