ClockSelect/myevic #132: pid algo

RenegadeTitan, 2016-11-04:
just started messing with this but based on what i have read about calibrating pid this may have been source of previously reported +35 degree (f) high temp regulation with default settings when in pid algo that i encountered. when doing trial and error calibration to p with i=0 and d=0 i am unable to get device to hit temp even with p=2000. it seems to reduce wattage before achieving temp regardless of p setting. even at max setting it slowly approaches below temp over 2-3 seconds but most importantly does not produce an oscillation. my understanding from reading is that with i=0 and d=0 p should be increased until oscillations begin to occur and then i is slowly increased to remove these. maybe p is missing x10 modifier? what is boost setting doing and is that a modifier to one or all of pid settings? maybe p is calculating based on (temp setting - x) to avoid overshoot? i have 30 minutes total of knowledge/experience with anything pid related so as a fair warning i do not know what i am talking about yet lol. 

i could not find discussion of your pid algo so was hoping you could help me understand your implementation of it so that i could use it more effectively.

ClockSelect, 2016-11-04:
The P-only can never stabilize or oscillate around the target temperature; this is a property of P regulation. It stabilizes or oscillate around a value with an offset below the target. It's the role of the Intergral part to fill the gap between the P offset and the target.
You're trying to use the ZN method; it may work but it takes some time as a trial-and-error method.
I used a more procedural method described [here](http://controlguru.com/table-of-contents/).
How does the default value behave with your setup?

ClockSelect, 2016-11-04:
In the doc, I give you the units of the parameters. As you can read, P in in mW/°C; it means that the PID will command a power of P * ( temp diff ) mW. If P=600 (default) and you want to up temp by 180°C, it will set the power to 600*180mW = 108W (at first). You may need more to get a fast ramp-up with your setting; go to the 250W and try to set P=1400. Increase I accordingly, so I=2000.
This should do the job.

RenegadeTitan, 2016-11-05:
still messing with pid settings. seems lower I and maxed out P gets me closest. adding small D seems to help. here are screenshots:

algo off:
[image]

default algo pid:
[image]

RenegadeTitan, 2016-11-05:
PID P=1400 I=2000 D=0 (boost 0%)
[image]
:

RenegadeTitan, 2016-11-05:
@ClockSelect i think i closed this unintentionally. if you closed it no problem. ill figure it out. just brought it up because it seemed like strange behavior. its most likely something im not doing right.

ClockSelect, 2016-11-05:
You keep your max power setting too low.
Using PID, you don't care what your max power setting are; just set it at 250W and forget it. The PID will just take what it needs. You're not leaving enough power to the PID to give you a good ramp-up.

RenegadeTitan, 2016-11-06:
yes that works much better. thank you. that was the issue.

for continuity with other vape modes, and increased versatility, in the code, shouldnt maximum mod wattage always be applied to PID equation regardless of set wattage? set wattage should only factor in as a ceiling, not affecting or modifying behavior of algo when device is regulating and firing below set wattage in my opinion. i believe this is how all other algo modes operate

for one example of benefit of keeping set wattage as only a celing: i use 2 battery configuration so mod has 200w max limit. but i prefer to keep batteries from firing at over ~20a to extend their life. i prefer to keep maximum wattage at 120w so that as batteries drain and series voltage approaches ~6v i dont have to worry about batteries applying too many amps. 

i dont believe this would negatively affect lighter coils. it may actually improve firing on light coils as it did for my coils. i am guessing it could require modifying default values but if your default values are results of calculation they may not.

ClockSelect, 2016-11-06:
Methods are different.

The Boost algo works similar to Joyetech's , except it considers different phases in the puff (ramp-up, stabilization, regulation...) and behaves differently in differents phases. The boost parameter is used to accelerate the ramp-up phase; the heaviest the build, the higher you should set it (maybe 80% in your case). It may work well in your case.
The Sweet algo is more or less a simplified zero Boost. It dampens oscillations on light builds.

For PID:
Don't bother with the **D**. It brings very little improvement if any.
If you limit the available power, you should decrease **I** to avoid the long overshoot you see on your curves, since the integral builds up during a longer time.
You'd better use 3 cells with your setup, and unconstrain the PID.

ClockSelect, 2016-11-08:
I added a battery protection in the PID algo; you can safely set your power to max and the box will adjust its max power depending on the battery state-of-charge. It will not draw more than the rated continuous discharge current for the selected battery model.

RenegadeTitan, 2016-11-08:
please forgive my slow posting in this thread, i am still playing with this whole area and require more time to develop productive ideas.

it is a good idea but i am concerned by any additional limitations that are not editable/adjustable. are the limits based on rated amp limits or tested amp limits? i have not seen a 18650 actually test over 23a and most/all 18650 will incur some level of damage at 20a even if "rated" at 25a/30a. also this would cause vapes to change as battery lowers.

i would prefer a less restrictive solution if we can find one. 

i am getting very different results depending on starting coil temperature due to differences in time required to reach set temp (same cause affecting the set wattage at 120w vs 200/250w different results, time to reach temp causing variances in regulation). it is my opinion that set wattage should be applied to achieve temp then regulation maintains it. what if there was a function that applies set wattage until actual temp reaches (set temp -x) then algo kicks in (x could be 10 or 20 or possibly settable by user)? this would make every vape very consistent regardless of starting temp or set wattage.

RenegadeTitan, 2016-11-08:
unrelated, just showing off the build :) SS316L twisted 24awg core wrapped in 30awg 4.5 wrap 5mm id dual core:
[image]

ClockSelect, 2016-11-08:
Applying the set wattage until reaching target temp is what the PID algo does if you give it a lower-than-max power; if you look at your curves, it's exactly what happens (except that there's some overshoot due to integral action, but that's tunable whith the parameters).

The high-current heating phase is very short if you give the algo the full power range; it's a ramp-up spike that lasts less than a second, and batteries can handle this kind of spike fairly well. I use Mooch's chart to define what the max Amps should be for a given battery model; he made reliable testing on batteries and my settings are not too conservative albeit safe for the cell. Anyway, at high Amps, you're limited by the internal resistance and the voltage sag that would drive voltage under the 2.8V limit, and thoses limits are close to each other so it's not really a limiting factor.

With near depleted battery, it's the 2.8V limit that takes precedence and the max Amps cannot be achieved anyway (the box would shut down), so there's no real concern with battery stress.

So you can safely set your power to the max allowed by the box and forget it; the algorithm will use what it needs and perform at its best.

Nice heavy build :)
I just ordered some SS staggered fused clapton to experiment with this kind of stuff.

RenegadeTitan, 2016-11-08:
thanks! i make the claptons myself with a drill and clapton rig i made out of a clothespin. but fused are 3x the work to make so probably better to just purchase them. i want to try those. i think they may work the best of any coil type. imo any ss coil is better than any coil of other material. enjoy :)

thumbs up to everything in last post, you relieved all concerns i had. good to know i can set watts to max and it will be fine.

i do still prefer to have the option for setting watts lower than max. i will try different PID settings for this. still experimenting. i do still think an option for adjusting the PID kick in point could be a good idea but now that i know there is a kick in point ill play around and see if that factors in much/at all.

RenegadeTitan, 2016-11-08:
i just noticed P max setting was dramatically increased from previous limit of 2000. that was one theory i had as to possible cause. thumbs up again. ill keep playing around.

ClockSelect, 2016-11-08:
**P** is in mW/°C, that means the controller will drive the power to P*δT mW, where δT is the wanted increase in temperature. So even with 250W available you should never need more than ~1400. Higher values won't accelerate the ramp-up and will increase instability.
Maybe it was not a so good idea to increase the limit; we'll see.

RenegadeTitan, 2016-11-08:
ramp up was adequate at 1400 but was overshooting. this was source of overshoot. it is now adjusting faster and not overshooting. device is still limited by wattage setting as it should so i believe calculating propper P setting based on max wattage is incorrect. P set too high will never exceed this. the concern shouldnt be that P creates too high a wattage at large distance from set temperature as PID does not function at great distance from set temperature. concern should be that P creates high enough wattage change at medium/small distances from set temperature to regulate fast enough. this seems perfect :) higher than 6000 may even appropriate.

RenegadeTitan, 2016-11-08:
you may be working on firmware but i wanted to mention that i dont see the bin folder.

ClockSelect, 2016-11-08:
bin came back :)
messed up during a pull request.

RenegadeTitan, 2016-11-08:
phew! all is right with the world again :)

RenegadeTitan, 2016-11-08:
i havent played with settings yet but compared with algo off its great. plus i was starting to get major oscillations intermittently with algo off so now i can stop troubleshooting that lol.
P=3000 I=850 D=0 :
[image]

ClockSelect, 2016-11-08:
That seems not bad at all.
You still limit yourself to 120W.
How does it behave if you unleash the horses?

RenegadeTitan, 2016-11-08:
same settings with 200w gave small overshoot but with quick wattage adjustments. very unexpected. lowered I to 700 and got this:
P=3000 I=700 D=0 200w :
[image]
i bet I~600 would be even better. have to pace out tests, too much nicotine lol.

ClockSelect, 2016-11-08:
Looks not very far from the best achievable :)

RenegadeTitan, 2016-11-08:
i agree. slope of line to set temp is almost consistent. as far as regulation at temp i think at perfect settings all bumps could possibly be size of first 2 bumps. but just an estimate based on accuracy of ohm readings. looks good so far :)

RenegadeTitan, 2016-11-08:
quick question, well may not be a quick answer lol, does set wattage affect any PID calculations that would directly affect temperature regulated wattage output?

ClockSelect, 2016-11-09:
Yes and no.
If lower the wattage, ramp-up will take longer, so error integral will have more time to build up and ultimately generate an overshoot and oscillations. So if you lower the wattage, you have to lower the **I** accordingly to prevent those adverse effects.
Beyond that, it has no influence.

RenegadeTitan, 2016-11-09:
ok cool. thats all making sense.

so heres what im thinking:
the temperature each vape starts at varies due to time from end of previous vape which would cause different PID calculations. you have minimized this with the PID only kicking in closer to set temperature after applying set wattage to reach this temperature point. so far so good.. but i am still noticing a variance in performance with more frequent vaping and a strong curve at the top of "preheat" phase that starts relatively low with some PID settings. i think raising the PID kick in temperature point even further (closer to set temperature) would further improve consistency for me. but this ideal temperature point may vary based on preferred PID settings (settings that are more finely tuned for maintaining set temperature but are more susceptible to warm up effects. i propose, if not too much trouble, making the PID kick in point (degrees below set temperature) adjustable.

ClockSelect, 2016-11-10:
The PID applies from start. There's no such thing like waiting for a target temp, or kick in point. Just, the max power it can apply is clamped so it's constrained on a straight line.
If it starts on a warm coils, it'll just apply less power to ramp up since Pwr=**P**xδT+**I**xΣδT, and δT is smaller. If you consider the extreme case, imagine you're firing while coil is already at target temp, the PID will apply just no power at all, and that's you'd expect.
Integral part does build up more slowly and has less effect on an already warm coil, so the convergence to target temp is slowed down, and maybe it's what you feel. You can increase **I** to counter this, but at the cost of a potential slight overshoot on cold coils. PID is always a question of compromises.
Best thing would be to increase the available power to 250W with a third cell, so that you can increase **I** without overshooting on cold coil, and keep a good behavior on warm coils. After all, your build is not really made for dual-cell.

ClockSelect, 2016-11-11:
I think the new profile feature will come handy when you'll have multiple atomizers with each of them having their own tuned PID parameters^^

RenegadeTitan, 2016-11-11:
cool. thats a nice addition. thank you.
also i appreciate that you take the time to understand my wordy, complicated posts and give such thought out and detailed responses.

RenegadeTitan, 2016-11-14:
i would like to mess around with with the algo functions (assuming you dont mind me experimenting) and try to impliment an idea im working on. i have little to no experience with c. but i was very proficient with hyper logo on the ti-84 a long time ago. i wrote some games and physics examples. c is actually pretty similar (at least structurally). but im getting really hung up on setting up the environment for compiling in windows. im following ReservedField's evic-sdk instructions and gleaming a little info from your notes in your makefile. any help in simplifying the process of compiling would be appreciated so i can focus my efforts on the code itself.

ClockSelect, 2016-11-14:
In fact I just followed instructions from eVic-SDK to setup my environment and it worked just fine. Made no change to the env since.
So I can hardly give you any advice without some more informations.
I suggest you to open a separate issue for this; so everybody can come to help.

RenegadeTitan, 2016-11-14:
great! thank you for clarifying for me that i can just follow those instructions. i will keep trying to fix errors as they come up and if i get a little stuck ill open a thread. i expect to open a thread soon lol.

closed 2016-11-16 by RenegadeTitan

sandbagjohnny, 2017-01-30:
@ClockSelect Thank you for this info! [I've posted an essay/mini-review of the new temp control methods on reddit](https://www.reddit.com/r/electronic_cigarette/comments/5ql2kn/on_pi_regulation_and_arcticfoxmyevic_some_notes/) based on both your comments here, and [maelstrom's comments on PI in his fork](https://github.com/maelstrom2001/ArcticFox/issues/91)

sandbagjohnny, 2017-02-14:
Thanks :) I've cleaned up the essay a little bit to reflect this, as I agree, both packages have their own unique style, and I've heard as many positive reviews of your offering as I have of AF. Thanks for your contributions!!