ClockSelect/myevic #133: Feature: Add "PPwr behavior" for preheat in Power Mode

phantasm, 2016-11-04:
In TC mode we have a great ability to quickly switch between Temp and Power limits (if PPwr mode is enabled).

It will be awesome if the same feature will be able to work in Power mode, switching between main power setting and preheat power (maybe with preheat duration, too). What do you think?

ClockSelect, 2016-11-05:
Why not.
It's question of interface design; I can't see how I'd do a user-friendly thing.
Maybe just using the PPwr click shortcut in PWR mode, if you defined one, sends you is the preheat menu? Simplest option.

phantasm, 2016-11-05:
@ClockSelect  that's exactly what I mean :) if PPwr click will switch between main power, preheat power and duration, it will be great.

ClockSelect, 2016-11-05:
That some code.
I though bringing you straight into the preheat menu.
Or maybe just switching preheat on/off would be enough?

phantasm, 2016-11-05:
@ClockSelect preheat menu will be good too, but it's harder to switch back to main screen, because you need to use +/- buttons and several clicks.

ClockSelect, 2016-11-05:
Fire-(+) gets out out of menus back to main screen instantly.

phantasm, 2016-11-05:
@ClockSelect oh, indeed :)

ClockSelect, 2016-11-05:
Try this.

phantasm, 2016-11-05:
@ClockSelect seems to work nicely on my RX2/3 so far

closed 2016-11-05 by ClockSelect