ClockSelect/myevic #135: Feature: Advanced Preheating

rsrmean, 2016-11-05:
Hi!

I built a Clapton Coil which needs some time to heat up. Preheating is fine, but once the coil is warm, it's not necessary anymore. Turning if on and off all the time is a hassle.

Maybe there could be another option in the Preheat menu, so preheating only kicks in if you didn't fire since XX seconds? I don't know if it's possible, and I don't know a good name for a feature like this neither :)

Regards

ClockSelect, 2016-11-05:
It exists, already; it's called Temperature Control^^

Nestorfish, 2016-11-05:
Factory coils are mostly kanthal and not supported by TC mode (I am in this case).

I tried multiple settings for pre-heating and did not settle on any, because they were inefficient to warm up the coil, or would dry the cotton once it is "hot".

Maybe a separate button press sequence to trigger preheat would avoid the need to store another setting.

rsrmean, 2016-11-06:
@Nestorfish very good idea. Just an addition to the click menu!

ClockSelect, 2016-11-06:
I added a dead time delay feature to the preheat menu.
This should do the job.

rsrmean, 2016-11-06:
It does indeed!

closed 2016-11-06 by rsrmean