ClockSelect/myevic #137: Interface lag?

phantasm, 2016-11-06:
When navigating through the interface, some strange things can occur - menu items are skipped, etc. Maybe it's related to board performance, but maybe it's a bug. It's definitely not related to button jitter -  I've verified it with various click speeds and thumb force.

Video: https://www.youtube.com/watch?v=YYXHNVZuBDk

Firmware: Any recent
Box Model: RX2/3

ClockSelect, 2016-11-06:
Buttons are polled at 5Hz; if you're clicking to fast, you're interfering with the polling frequency and the box goes into key repeat mode and accelerate, thus going faster than the display refresh rate (10Hz max).

phantasm, 2016-11-06:
@ClockSelect maybe increase polling rate when in menu mode? It won't eat too much battery (because you don't open menu too often), but will be a lot smoother.

RenegadeTitan, 2016-11-06:
i have experienced same behavior. if  polling of 10Hz is possible this may allow for fast clicking. i dont think more than 10 clicks/second will be common enough to be an issue but i belive 6 or 7+ clicks per second is relatively common even in menu navigation.

ClockSelect, 2016-11-07:
Let's test 10Hz.

phantasm, 2016-11-07:
@ClockSelect now it's much, much better, thank you! hope that polling is increased only in menu mode and will not eat battery.

ClockSelect, 2016-11-07:
This kind of change does not impact battery consumption at all.

closed 2016-11-07 by ClockSelect