ClockSelect/myevic #139: PreheatDelay masks under-load battery voltage display

MarkyAD, 2016-11-07:
Box Model: VTwo mini/VTC mini

Without PreheatDelay, it is possible to see the under-load battery voltage just after taking a puff. When enabled it masks displaying it, so that by the time the display gets refreshed, the battery has already returned to its no-load voltage. Thanks for all the updates, features and fixes.

MarkyAD, 2016-11-07:
Its not that the display isn't refreshed soon enough, but rather too soon. If it waits until the onscreen puff timer disappears, that gives enough time to see what the voltage under load was. Increasing the value in timers.c which tests for ! (--PreheatDelay % 25) avoids the issue but then the P doesnt blink often enough.

ClockSelect, 2016-11-07:
I see.

ClockSelect, 2016-11-07:
Fixed.

MarkyAD, 2016-11-07:
Nice work once again. Fixed and closed.

closed 2016-11-07 by MarkyAD