ClockSelect/myevic #140: Battery Mode for VTC5A's?

LordNex, 2016-11-07:
**Box Model:RX200S also have Cuboid but this is related to RX200S 

**Coil setup: Several but for instance Dual Kanthal A1 28/32 Fused Clapton 5 wrap on 3mm @ 25ohm

Quick question. I just got a set of VTC5A's in and I noticed you had custom battery profiles for VTC4, 5, & 6's but no 5A's. Mine are Authentic 2600mAh 30-35A Sony VTC5A's. Should I be running those under the VTC5 profile or do you suggest another one?

Also just want to say thanks a lot. A guy posted this to Reddit a few weeks ago and have been following ever since. I have your firmware in both my RX200S and Cuboid and love it. I have a bunch of questions but I'll post them later. Again thanks and keep up the kick ass work. It doesn't go unappreciated!

LordNex, 2016-11-07:
That should say .25ohm BTW

ClockSelect, 2016-11-07:
Thanks for your comments^^

I don't have the charge curve of the VTC5A, so I can't really tell.
VTC5A is just a chemistery tweak of the VTC5, so they should be close, in theory.
If unsure, use the generic battery setting.

bluenazgul, 2016-11-07:
Not sure if it helps, but I found one for the Sony VTC5a

http://dna200-themes.de/wpdm-package/sony-vtc5a-therion/

ClockSelect, 2016-11-07:
Thanks!
Yes, they're very similar.
Yours is a bit more optimistic on the low side, but voltages under 3.1V are useless on Joyetech's chipsets.
VTC5 curve is fine to use, IMO.

LordNex, 2016-11-07:
It seems like when I select the VTC5 profile the battery bar drops a small bit. I only noticed it because I had them on he charger all night and the profile still at generic. I put the batteries in, bar was full, then changed profile and there was a small gap maybe 1/10th or less in the battery bar. I didn't have % on at the moment so I'll have to test that more after work but thought I'd let you know. Again thanx for everything!

ClockSelect, 2016-11-07:
If you choose a different profile, it's perfectly logic you get different numbers.
The generic profile is optimistic; it shows high percentages on the run and drops quickly at the end. It tends to always over-estimate the state of charge.
Also, bar only shows 10th rounded low, so 99% is drawn as 90%.

LordNex, 2016-11-07:
Gotcha, makes sense.

closed 2016-11-08 by ClockSelect