ClockSelect/myevic #141: Make "Battery Low" limit adjustable?

dabugen, 2016-11-07:
Is there a way to make "Battery Low" limit adjustable by letting us set the lowest voltage allowed for the battery? I am using a Presa TC 100 and use a 26650 battery which, if at about 3.4 volt, drops down to 2.85 during firing for just a few moments but then returns quickly to 3.1 volt. The battery % shows 25% left, but I can´t fire anymore most of the time because the "Battery Low" message kicks in and prevents the vape. Is there a way that you can let us adjust the lower voltage limit for that message in the Experts menu (at our own risk of course)?

Thank you.

ClockSelect, 2016-11-07:
Your numbers mean that you vape at around 50~60W.
There's almost no gain to expect by lowering the cutoff by a fraction of a volt.
3.4V is somewhere between 5 and 10%, not 25%; the generic battery curve is way too optimistic.
The best solution (by far) to your issue is a second battery and a battery charger.

dabugen, 2016-11-07:
Yes I know, this is the ramp-up for the heat-up during the initial TC-spike that causes that voltage drop. However, can you still make it adjustable in the experts options? Because the voltage drop only kicks in for a split second and then everything is fine again and the vape is great (if trying to fire several times). I just don´t want the "Battery Low" message to cancel the whole vape in such a case, hence the request for making it adjustable (at my own risk of course).

ClockSelect, 2016-11-08:
Battery management and cut-off thresholds are the primary reasons why I went into this project.
I had the very same concerns of wtf? 25% left and I'm locked and the specs of my cell says 2.5V is fine. So, I made my firmware just for that. And finaly, I didn't change anything, because things are well done on this aspect.
The hardware fails at around 2.7+V and it's a hard barrier.
The firmware stops fire at 2.8V to avoid hardware failure.
Then it regulates fire at around 2.9V to give itself a safety margin.
It refuses to wake up from power-off mode below 3.0V;
It refuses to fire below 3.1V, and locks fire unless it reaches 3.3V again after beeing below 3.1V.
There is no limitations otherwise.
It tried to play with thoses numbers; always caused more problems than I solved.
3.3V is an empty battery.
Just swap it.

dabugen, 2016-11-08:
As I said, start of the vape it is at 3.5v, then just drops for a split second to below 2.8 and it cuts off the whole vape, makes no sense. 3.5v is absolutely fine, my 26650 can be taken down to 2.5v as per their specs and I am using them in other devices down until that voltage. Apart from that, it really recovers quickly upwards again, it just is a first drop that goes low (the brand I am using is known for that, but still is great). Re-charging at 3.3 or 3.5v is surely not a empty battery, that is nonsense, because those can be used in other devices just fine down to 2.5v without any issues (drones, etc.). Can you hence please just make it setable where the cutoff happens - in the Expert menu it should just fine. Thanks.

ClockSelect, 2016-11-08:
You fire your drone at 60W?

dabugen, 2016-11-08:
No, hence the temporary voltage drop is not that high, but those batteries still do fine in there at 2.5v, so neiter 3.5v nor 3.3v mean that the battery is empty, that is just non-sense. Can you please make the limit adjustable? What´s the big deal with that?

StepWi, 2016-11-08:
The big deal is that the hardware can't be "adjusted". If it stops firing at 2.7V or has unreliable behaviour, there is nothing that could be done with software parameters.

dabugen, 2016-11-08:
OK thank you, if this is a hardware lock, then it can´t be changed indeed. But at least the software could be brought down to 2.7 as well instead of 2.8.

ClockSelect, 2016-11-08:
The harware shuts down and reset around 2.75V; with the inaccuracy of battery readings (there's some conversion noise), 2.8V is the best it can do. I already tried to lower it, had box reset issues.
If you want to experiment, you may try to fake the voltage with the Expert>BVO feature; offset your battery voltage by +0.1V, and this will give the exact same result as lowering the cut-off voltage.

ClockSelect, 2016-11-10:
New: you now can use a custom battery profile, and play with the cut-off value by editing the battery profile with the NFirmwareEditor.

dabugen, 2016-11-10:
Oh great, how do I access that editor there? Did not find it...

dabugen, 2016-11-10:
Cool, how can I access that battery profile editor? Did not find it yet....

ClockSelect, 2016-11-10:
Menu myEvic NFE Edition > Device Configuration > Developer Tab > Expert Tab, choose "Custom" in Battery Model drop-down list, then click "Edit".

dabugen, 2016-11-11:
Thank you, but I am always receiving: "Waiting for device with myEvic NFE Edition 161002 or newer"....

ClockSelect, 2016-11-11:
Use the NFE option in the Expert menu.

SEEKNDESTROY, 2016-11-11:
Also, make sure your mod is not in sleep mode.

dabugen, 2016-11-11:
Thanks, works now:)

ClockSelect, 2016-11-11:
Great :)

closed 2016-11-11 by ClockSelect