ClockSelect/myevic #142: Which battery slot is which on RX200s?

SIimJim, 2016-11-07:
**Box Model:** RX200s

Hi Guys Amazing work!!!
Quick question I am trying to sev the correct voltages for my batteries but I am getting differnt readings in device (RX200s) compared to  my multimeter and I just need to know which slot is which, my instinct tells me looking at the back of the device the left inner battery is 1 right inner battery is 2 and the center outer battery is 3 but things just don't seem to add up please advise :) thanks again for your amazing work I literally LOVE it :)

bluenazgul, 2016-11-07:
Seen from Display Side:

   3
2   1

SIimJim, 2016-11-07:
ok great just like I thought :)

ClockSelect, 2016-11-07:
@bluenazgul  Hm... Seems that's the layout of the RX2/3.

RX200S would be:

 2
3 1
---
^display

bluenazgul, 2016-11-07:
@ClockSelect This is the only Layout i found for the RX2/3 - I think RX2/3 and RX200S should be the same, needs to be verified with two full charged and one empty Battery i think

[image]

SIimJim, 2016-11-07:
You know what that is such an easy way to test it is feel dumb haha I will have to go get my Koopor Mini out the car to run a battery down :) I will post results asap

ClockSelect, 2016-11-07:
That's what I did, once^^
But just by looking at battery polarities, you find out the current path and wich one is wich.
The rearest battery in the RX200S is the number 2.

SIimJim, 2016-11-07:
So this is confirmed in the RX200s?

2
3 1
---
^display

ClockSelect, 2016-11-07:
Yes.

ClockSelect, 2016-11-07:
Btw, if you short the connectors of the third battery (left inner), the RX220S will perfectly work as a dual-cell box just like the RX2/3 (tested, works). It's a bit useless since you have to keep the number 2 and thus cannot reduce the size of the box, but it's a curiosity.

SIimJim, 2016-11-07:
Haha cool, well thank you very much for the help and myevic in general 👍

closed 2016-11-07 by ClockSelect