ClockSelect/myevic #144: Suggestion: advanced manual power "throttle"

Derducken, 2016-11-08:
From what I read, since I don't have such a device, both the DNA chipsets as well as the latest mod from Aspire allow the user to set "power fluctuations" during the duration of the vape. It supposedly works in the same style as the Preheat already supported in the MyEvic firmware, but instead of setting **two** points where the power will change (Point A when you first hit the button - the "preheat" - and Point B after X seconds) you can add more.

Do you think something like that would be feasible to add?

falconettt, 2016-11-08:
I'd like to support the issue.
@ClockSelect : I've tried in #59  to suggest something like this but you refused.
It's all about power mode.

Power modulation of some kind is achievable now in TC mode, but does not improve the taste and leads to heating of the atom (IMO).

RenegadeTitan, 2016-11-08:
i understand the thought that a quick cyclic progression of temperatures (not just power) would produce a spectrum of flavors corresponding to temperatures. but because taste receptors respond so slowly i would think a steady progression through the same temperatures would produce the exact or nearly exact same flavor spectrum. also because increasing temperature (every pulse/cycle) is less efficient than maintaining temperature, this would produce **much** less vapor for every watts/s.  i agree with ClockSelect that minimizing modulations (in any mode) is optimal. this is just my opinion and i could be missing some facts.

falconettt, 2016-11-08:
@RenegadeTitan : original post is a a special case of the general process  of low frequency PWM. May be one point or may be ten of them. It needs testing.

One thing is clear: VW as it is a bit boring. One wave of preheat makes it a bit interesting.
May be more waves of power will make it even more interesting?

ClockSelect, 2016-11-08:
Did someone actually tried this feature with a capable box?

Derducken, 2016-11-08:
I'm not thinking about TC - but if we had at least a THREE-point level of control of VW mode, we could better aproximate TC with Kanthal. Lemme explain my reasoning (at least, based on how *I* vape).

We set preheat for 50W and 0.5 secs for our coil. We set our power at 40W. VW mode, Kanthal. We hit the button.

The function of "preheat" is to actually lessen ramp up time - to reach "X" temperature faster. So, let's say everything's perfect and we, indeed, hit X temperature. AFTER preheat, the mod switches to Y temperature, our "normal non-preheat Watts setting". Supposedly "to keep the temperature steady".

Thing is, there's a constant drop in temperature felt if you take long drags. You have a burst of energy when starting (the preheat), then the temperature drops while a "steady current" is applied on the coils (the "normal" power level), then that's it. Whatever you do, the feeling of the vape WILL get progressively colder.

If we had an at-least three point system, we'd be able to set preheat at 50W, a second "point" at 40W and an "endpoint" at 45W, so a constant, steady increase in power would keep the feeling of the vape at a similar temperature.

Of course, that's a specific problem applying to a specific way of vaping. I like long, over 5 seconds puffs, so I *can* feel the vape getting progressively colder. The only way to avoid it is by setting the "main power level" higher, but then there's a point between the preheat and the "normal vape" that can feel a bit warmer - harsher than what I'd call comfortable (it last less than a second, but it's there).

Anyways, it's just an idea :-)

ClockSelect, 2016-11-08:
If you look at a typical TC curve:
(Power green on top, temp red just below)
[image]

The power first stick to the max TC power setting (here, 20W) that acts like a preheat, then the temperature and the power both stabilize steadily to their respective cruise values (~15W and 180°C).

The temperature does not drop with time, and there no hot spot anywhere. If I set a 0.3s 20W preheat followed by a 15W main power in wattage mode, and tweak the values a little, I should be able to reproduce almost the same curve without the need for a third point.

StepWi, 2016-11-08:
@Derducken
If you would give more than 40W, the vape wouldn't become colder. If the first hit with 50W is to hot, than reduce the preheat time. From your values given in the example, I would raise preheat to 60 up to 75W and reduce preheat time to 0.2 or 0.3 seconds.
And you need to inhalate constantly to keep temperature constantly. If this does not work for you, try TC mode (with VA-wire) as ClockSelect suggests.

ClockSelect, 2016-11-08:
The point is about the power curves, did someone really tested them?
I searched some reviews but couldn't find any.
ATM, I'm 99% convinced this feature is almost useless.
But I remain open.

afol2, 2016-11-09:
Back in my days... i used to have an Evic, the long tube capable of 11W max, it has a pc software in wich you can set a time and power, i think it was the beginning of "TC". for example... 0.1 @ 11W, 0.3 @ 8W and then change at 1 sec @ 5W, and it used to have like 10 slots of variable wattage back in the days, this feature was usefull, as it was almost everything genesis style atomizer and you can push it to give it more flavor and vapor without the dryhit another cool feature it used to have, was that you can select like 3 atomizer that was storage in the memory, and for example I used to have an AGA T+ atomizer, i select that Ato, and it recall the power curves features I input for that atomizer, i think it used to have 3 slot for it. 
[image]

ClockSelect, 2016-11-09:
I had this eVic, with a ProTank II in those good'ol'days. Still have it somewhere, btw.

Those features has been discontinued in more recent devices because of Temperature Control.
TC made power curves useless.

falconettt, 2016-11-09:
But VW with preheat is a a special case of power curve. It came after TC and nobody say that preheat is useless...

falconettt, 2016-11-10:
> Did someone actually tried this feature with a capable box?

I did this by pressing the fire three times:
[image]

And YES, my experience is much better then this:
[image]

And btw:
Your contribution to the flexibility makes your firmware attractive and not your contribution to deciding what is good and what is bad. Leave the possibility of a decision to the user.

RenegadeTitan, 2016-11-10:
@falconettt ClockSelect is the one that has to make the changes. it is up to him. in my experience he has been very reasonable in discussing why a proposed change is an improvement or not. his ability to rationally choose what to put in and what not to put in (not just adding every suggestion) is what has let to this firmware being as great as it is.

falconettt, 2016-11-10:
@RenegadeTitan : I do no come in the room with your "pid algo" issue just to say "Hey, I'm not interested and not convinced". I respect you and do not want to spoil your game.  So please respect me and do not spoil mine.

RenegadeTitan, 2016-11-10:
my comments are always meant to be helpful. i saw your last line requesting that he "leave the possibility of a decision to the user" and believed this may not be the most productive type of comment to convince him. i also saw that your screen shots are not comparing the most useful scenarios to prove your point. i made my last post entirely to help you:
A: have the correct understanding of your position when making a proposition.
B. provide the most useful data to prove your point.
i dont use vw mode so do not have an interest in if this is implemented or not. i am not trying to stop this change. just contributing to the discussion.

Derducken, 2016-11-11:
@falconettt You don't "get it": I asked **if** it was possible for a feature **I** like, and thought others like you would find useful. Notice that I **asked**. I didn't **demand**.

The way your message was written made it look like you **demanded** ClockSelect add this feature. It's not our place to tell him what to do, 'cause it's, notice that, **HIS** firmware, **HIS** work. He **SHARES** it with us. He could develop it and keep it for himself. We should be glad we have what we have thanks to him.

@ClockSelect I don't understand how preheat curves can replace the feature I mentioned. As in, I still don't understand how exactly the new curves feature of the firmware works. When I enabled the "sweet" one, preheat seemed to have stopped working - when I fired it, my VTC Mini didn't mention a higher Wattage while, theoretically, it should (since preheat was selected and set at 0.5 secs). So, do curves disable preheat and replace it somehow? Did I do something wrong? If that helps, there was a tiny "p" blinking on the top right of the reported Wattage.

Ah, and on a completely unrelated note, thanks for your reply regarding the fonts. You "closed the issue" before I found the time to thank you for your reply there - I'll try to inject some other font in the firmware myself with the tools you mentioned.

Am I allowed to... er... pester you with some more ideas?!? Should I mention them as different posts/tickets (for easy reference), would it be easier to create a "mega-ticket" with some of them, or should I simply forget throwing my ideas around and shut up? :-)

ClockSelect, 2016-11-11:
@Derducken 
Ideas are always welcome. Open one separate issue for each different idea so we can close/let open/discuss each of them separatly.

The preheat function is a Power mode feature, while Algos are Temp Control features. Algos have no effect in Power mode, and Preheat has no effect in Temp Control modes.

The blinking "P" means you just draw a puff and you're in the dead time delay configured in the Preheat menu. The preheat won't re-heat the coil while it's still warm before the configured amount for time after last puff. That's why you don't see the higher wattage when refiring.

Derducken, 2016-11-11:
Man, that's awesome, I didn't know the preheat worked "selectively" like you describe!

I thought the curves were replacing the "simple preheat" function with "something more advanced" and had thus disabled it in my case to do something *I* couldn't actually feel. Thanks for the clarification.

falconettt, 2016-11-11:
I understand that demands are inappropriate here.
I tried to be persistent and perhaps overdone. My apologies. 
Maybe it is the difficulty of automatic translation to English.
I wish you success in applying your suggestion.

Derducken, 2016-11-11:
@falconettt Oh, man, don't feel bad about it - English is a second language for me as well. I didn't want to bash you, just to remind you that we can *ask* for stuff, and if he likes our ideas, we *may* get stuff. That's the way it goes. If we don't like it, we can either *pay* him to add features ('cause why would he "work for others", to add the features *we* want, without some kind of compensation?) of learn coding ourselves.

ClockSelect, 2016-11-14:
Try this.
Documentation is still minimal, but you'll find out^^

falconettt, 2016-11-15:
Yessssssssssssssssss! 
Seems working OK. There is no need for documentation, everything is clear without. One thing: no exit when adjustments done (it is not critical). And strange behavior of 3-D saver, by the way.
Definitely going to try in detail, maybe I'll find something "unquestionable" there...
Many thanks!

But what about our "little steam locomotive" (overregulated PID)?
As to speak seriously, it could be an alternative of twisted coil or clapton, because it makes vapor more "velvetty" (IMO). This mode spoils the density of vapor but it is easily corrected with help of Ethyl Maltol. Possible needs: P>6000 (I don' know is it safe) and a bit of noise (I know sure it's possible).
You referenced and closed the issue. Does it mean or not you don't want to discuss it anymore? Please make it clear. I do not insist and happy as it is.

Derducken, 2016-11-15:
This is perfect. Or, at least, one step before perfect *for me*, since I'd prefer having something like a "hollow bar" for every "line" of the settings, with a vertical line in the middle @100%. I mean, looking more like "a bar with a slider in the middle" than a half-empty battery.

But that's just appearances :-D

It works and it's awesome! I'm vaping at 120% - 110% and then dropping to 90% and rising, step-by-step, one-by-one. It feels like a more "steady" temperature than with any settings I tried with preheat. Thanks. Really.

Derducken, 2016-11-15:
Another suggestion, for further improving this feature in the future (if and when you decide you've got time for this) : "Overheat protection".

I'm basically talking about a copy-paste of the "delay" function you use in Preheat, so that it also applies in the Curve settings, **but** only "ignoring" anything over 100%. Thus, if the curve starts, like in my case, at 120% but at some point goes to 100% and lower, with "overheat protection" enabled, anything over 100% would be ignored for X amount of time.

Even like it is, though, I feel like I'm using VTC 2.0 right now!

Some days ago I created a post at Reddits ECR regarding your firmware. You'll allow me to go rant there once more, this time about this awesome, unavailable-in-most-low-to-middle-end-mods feature you've just added. You rule dude!

ClockSelect, 2016-11-15:
I rapidly made something to edit a curve, but sure that with tree buttons it's a bit hard to do whatever we want. It'll be the job of a future configuration editor, and I have some hope we'll do something great with another team.
Remember that Apsire have five buttons^^ They're cooking with gas.

@Derducken : 100% means nothing to the curve. You may choose to be always above, or always beyond, or always in-between. Or choose to vary. I choosed to make the curve in percentage rather than in absolute power because you can scale it as a whole, and eventualy use the same curve in different power range. So "100%" does not mean a lot.

@falconettt : I moved the discussion here not to keep to issues on the same subject. Discussion is still opened... here :)
PID is not aimed to this. PID is a temp control algo, not a power algo. What Temp Control people like is, *Control*. Like "Control" in *Temp Control*.
You, you like out-of-control power. You have it, know^^

The look-and-feel of the inferface may change in the future, but for know let's keep it and see.

Derducken, 2016-11-15:
@ClockSelect Yes, I know, what I said was in context with chain vaping: when you *don't* use Curves, but the normal Preheat, you have a "delay" that sets a time during which the Preheat isn't applied for an X amount of time if you chain vape.

With the new Curves feature, when you, for example, set the curve starting at 120% "as a preheat", if you chain vape, you re-apply 120%, again and again. If you were using the "normal" Preheat, "the preheat wouldn't be applied because of the delay". *That;s* what I'm talking about, an option (maybe sometime in the future) that will regard "anything over 100% the same as the Preheat value in the normal preheat menu" and offer a similar option *not* to *re-* apply anything over 100% over X amount of seconds.

ClockSelect, 2016-11-15:
I pefectly understood what you meant :)
"Delay" is hardly applicable to the curves. Let's wait and see.
When you'll be diappointed with curves altogether, problems will solve by themselves.

Derducken, 2016-11-15:
PLUS: An explanation of how I use the feature that may help you understand why I found it useful - as I also explained on a Reddit post regarding this new feature:

I like long direct-lung draws. Thing is, when you don't do TC, and set your power so that you can easily take long draws, the coil takes a long time to heat up, your vape feels "wimpy" until sometime in the middle.

Now, if you apply preheat, you're trying to "make the coil heat up faster" and then, by using the normal power setting, keep the vape "steady".

Thing is, when you take looooong draws, sometimes this "steady" feels a bit... dunno... harsh after 3-4 seconds. I find that, depending and on atomizer, coil, wicking, sometimes I can easily take 7-8 second drags, othertimes I can't stand anything over 3 seconds,

Theoretically, a "middle-heat" would help in that, allowing me to lower the power a bit during the middle. And you provided, by implementing a full curve.

I know that it sounds strange in a world where people are used to taking quick 1-2 second high-power puffs, but I prefer lower-power long puffs, and this feature allows me to "customize their feeling" exactly. If I were to describe it, you can feel some points *when* you're vaping where you think "ah, it would be awesome if now it felt a bit colder/hotter". Now you can tweak this, at this exact moment. And repeat the experience.

It's awesome!

ClockSelect, 2016-11-15:
I'm ok with that; what I say is: first, play with it, *then*, I mean at least in a few days, tell me about possible improvements. Don't rush.

ClockSelect, 2016-11-15:
@falconettt : Exit always follows the same rules: Fire-(+) or (-)-(+).

SEEKNDESTROY, 2016-11-15:
So far so good, I'm kinda liking the Curve feature. I don't vape above 100 watts and I don't inhale for longer than 5 sec. I like to start with a warm vape and end with a cool vape. Below is what I'm playing around with.
[image]
85%
30%
60%
20%
35%
20%
35%
30%
35%
30%
35%
[image]

This feature would be perfect for people with crazy builds that only do 1-3 sec. inhales by setting everything to 0 after X number of seconds.

Thanks for everything you do!

ClockSelect, 2016-11-15:
@SEEKNDESTROY : "Protec" feature. Set it to X seconds, you're done.

SEEKNDESTROY, 2016-11-15:
Cool, I've never looked in to that feature since I don't do real long or short inhales.

ClockSelect, 2016-11-16:
Explore. You'll be surprized.

closed 2016-11-16 by ClockSelect