ClockSelect/myevic #145: Multi logo features

angah83, 2016-11-08:
Hi, can the firmware store more than one(1) logo? if possible, can you please add a feature/menu to select between them. two(2) logo would be suffice. thanks!

MicroSur, 2016-11-08:
And slideshow them )

bluenazgul, 2016-11-08:
A second Logo (hard themed inside the Firmware (there are two ;) one for each theme)) would be ok i think,  but a slideshow would be not worth the code i think

angah83, 2016-11-11:
@ClockSelect I saw that you have implemented the Profiles. Multi logo will surely get benefit from it. If I have two rta's eg; Kayfun & Merlin and I upload Svoemesto & Augvape logo, I can then assign my attach Kayfun atomizer with Svoemesto logo and Merlin with Augvape logo. Whenever I switch between the two atomizer, the logo will also change accordingly (taking into account that both atomizer have different resistance).

Villista, 2016-11-12:
Perhaps not separately downloadable logo, but the icons that can be edited in the resources editor (NFirmwareEditor or VTSFont) as the picture clock face. It would be cool!