ClockSelect/myevic #146: Request: another change of font

Derducken, 2016-11-08:
It's not an important problem for most people, but I'm 40 years old and my eyesight's already fading after 30 of them in front of computer screens. Yeah, I'm talking about the readability of the font. As in "I can't see what's on the screen without glasses".

If it wouldn't be too hard, could you add a third font, smack right in the middle of the other two? I mean, less-rounded, more-rectangular, but at the same time bold-ish? When your eyesight's ain't perfect, text with more corners seems "blob-y", and you can't clearly make out text that's "too thin" - so, both the current options end up unreadable for me.

Again, sorry for the request, as I said it's not really important, or a problem, but IF it's possible to keep it at the bottom of your "possible to-do's list", I'd be grateful.

Thanks.

ClockSelect, 2016-11-09:
Fonts take way too much ROM space.
Adding one is impossible due to memory constraints.
I may even have to *remove* one in the future to free some space for more usefull things.

ClockSelect, 2016-11-09:
Tools exist, like NFirmwareEditor or VTCFont to edit or import the resource pack you'd like; if you fond a font that's suitable for you, you may use those tools to replace the default font.

closed 2016-11-11 by ClockSelect