ClockSelect/myevic #147: Suggestion: user-edited "profiles"

Derducken, 2016-11-08:
Aaaand here's another idea - suggestion: how about being able to save EVERYTHING in profiles? You already support auto-loading profiles the mod creates itself for different atomizers (if I'm not mistaken). It would be trully awesome to have some preset slots in a menu where you'd select to save everything in profiles A, B or C, or whatever name (or placement) you'd like. And, by everything I mean everything apart from the resistance - preheat time and curves and all the rest, WW in power mod, temp in TC, up to screen brightness and logo on/off.

ClockSelect, 2016-11-09:
Problem is memory space.
Dataflash parameters are 256 bytes and almost full.
I already though about this idea; I have to find something smart to implement it.
Saving non-vaping related parameters is useless, imo.

Derducken, 2016-11-09:
Hm, I suspected that there could be such a problem. And I agree some values would be irrelevant - I mentioned "everything" since it's easier explaining to others "everything is saved in a profile" compared to "option A from menu B and options C-D-E from menu X, and blah-blah-blah, are saved in a profile".

Villista, 2016-11-11:
It's a great idea, and I understand how difficult it is to implement it! I've programmed microcontrollers and know how little memory! Can leave less profile in the mode of power for the atomizer? Can you use binary-decimal code storage? All logic On/Off can be stored in bits in flag bytes? Do not store at the same time, the percentage and power, and to convert one into the other! There is also the possibility to configure the profiles in the editor (like DNA or YIHI) and record it in the memory where the picture (there is enough space!).
Really really want the profiles under each tank or atomizer! Screw it, switched, and vaporize!

ClockSelect, 2016-11-11:
Made what I could.

Derducken, 2016-11-11:
Hmmm... The "save it where the picture should be" could be a possible solution to this. As in, the firmware does have a bit of "fluff" as it is (rotating cube / screensaver, game). *IF* something like that was removed (and if that's even possible), would it make enough room for saving/loading "profiles" in simple datastreams? As in, a simple txt file with a "010011100101001", where each 0 or 1 would (obviously) correspond to a specific "on/off" option.

I, for one, would prefer profile support on a mod instead of a rotating cube screensaver :-)

(...but that's me :-P )

ClockSelect, 2016-11-11:
Did you give a try to the last commit?

Derducken, 2016-11-11:
Er... Nope?! Damn! You've added it already?! I thought the "Made what I could" comment translated to "what's in the firmware's all I can do, sorries"!

Now I've got to reflash! Awesome!

ClockSelect, 2016-11-11:
ATM, it's made a very naive way; everything is reloaded, including transcient settings - contrast, stealth mode, etc. I don't think it's ideal and I think I'll add some degrees of filtering.

Daveee10, 2016-11-11:
IMO this things enough:
-Mode
-Power or temp
-Preheat values
-Resistance if locked
-ml/kJ
-Protection time

ClockSelect, 2016-11-11:
Should be fine.

Derducken, 2016-11-11:
Er... Am I doing something wrong? I downloaded the file from [here](https://github.com/ClockSelect/myevic/blob/8642da4f30fe76ed99541a5346d8ebaa62c7a2c0/bin/myevic.bin), flashed and have no profiles menu.

On the other hand, the mod's flying! Did you do anything to make the menus faster?!

Derducken, 2016-11-11:
On another note, saving EVERYTHING in a profile could be even better in the long run compared to "only vape related options". Possible usage scenario: two profiles, one for when "outside", one for "in-home use". Bright screen + stealth mode on VS darker screen.

I mean, since the user doesn't have to manually set everything up in each profile, "just change the things he wants different in the existing one and save it in a different slot", why not leave everything and the kitchen sink in, since you can? It's not a bug, it's a feature! :-P

ClockSelect, 2016-11-12:
What do you think I did?

Derducken, 2016-11-12:
Well, after Daveee10's comment, and you saying in the documentation that "vape-related stuff is saved" (if I'm not mistaken), I thought you saved only specific settings.

ClockSelect, 2016-11-12:
Before going any further, first, use it, or not, and when you'll find it's finaly useless, let me know.
If the purpose of this feature is, for you, tou switch your stealth and contrast settings not to disturb your cat when going back home at night, I'm afraid we don't agree on the purpose on an e-cig.

StepWi, 2016-11-12:
I preferr the "Everything" - Solution. It helps with
- different atomizers
- inside / outside - profiles
- battery change if you use 30Q and VT5

Villista, 2016-11-12:
Cheers! Finally there user profiles! It is the perfect solution! Tested in two tanks. For a single tank created two different profiles, and the third for the other tank. A little delay (or a bug) when selecting a profile short click "Fire" in the menu on the RX 2/3.
Maybe save the current profile when switching to another (autosave, which can be on / off in the menu)?
Maybe you can add switch profiles to the menu "clicks" (as cyclical mode switching) to select it on 2/3/4  click? Maybe it will add convenience.

The rest is just great! Many thanks to the authors!

Derducken, 2016-11-12:
@ClockSelect Hey, it was YOU who said "it saves EVERYTHING". Why the "attacking"? I just ask for a clarification of how the function works: I said it would be better if it worked like you said you had implemented it in the begining, "saving everything", and you answered...

"What do you think I did?"

I pointed out that in the documentation you say it DOESN'T work like this and DOESN'T "save everything", and now you're saying "we don't agree on the purpose on an e-cig"?!

Please, do inform me: did I demand anywhere you implement a feature that won't disturb my cat by blinking lights? You said you added a feature that worked "this way", the documentation said it worked "that way", and I simply asked which of the two ways is actually the one implemented. Don't get mad at ME for not knowing that when you say "yeah, it saves everything" here (like when you first implemented the feature) you DON'T actually mean "everything".

ClockSelect, 2016-11-12:
@Derducken misundertanding, no attack there^^

So yes indeed the box saves everything since when using a flash memory page, you have to use all or nothing, so you have space to save everything. This will permit for future evolutions if needed.

ClockSelect, 2016-11-12:
@Villista : the way it's done now aim to minimize flash memory write cycles, that's why I didn't made profiles cycling available on clicks. And cycling through five profiles is not really convenient. The fire+ is simple and easy, imo. And if you cycle, you take the risk to erase all resistance by the current resistance, that's not what you want either.
The delay and screen blink are normal too.

Villista, 2016-11-12:
Thanks for the explanation.

Derducken, 2016-11-12:
Hmmm... Another silly idea: would it be possible to "map" profile selection to the "multi-click shortcuts" of the fire button? Like, three clicks and, blam, next profile activated.

ClockSelect, 2016-11-12:
That's we just discussed above.
Well it'd be doable, but without saving the current profile - just switching to the next.

Derducken, 2016-11-12:
Yeah, I know - I understood what you said as "the shortcut function overtakes other clicks". I'm talking only cycling through them - like you cycle through "usage modes" (power, TC, Bypass). And ONLY if it's easy for you - no reason you spend time on fluff you, yourself, wouldn't find useful.

Villista, 2016-11-14:
I said in the post above, it is possible to add the profiles are not in separate menu, but the menu: Vayping-> Modes.

ClockSelect, 2016-11-14:
@Villista This wouldn't be as an easy access.

A multi-click cycling feature has been added.
Maybe some will find it convenient.

SEEKNDESTROY, 2016-11-14:
@ClockSelect very convenient... Thank you!

GeorgeEdward, 2016-11-14:
I think it would be better and more convenient to add in "Interface->Clicks" option not "Cycle through configuration profiles" but "Enter the Profile selection screen". In this case one can simply choose a suitable profile by +/- and by pressing Fire button switch to it. Isn't it?

Villista, 2016-11-14:
Oh! I'm probably too bad I can explain that very clear to me! I meant to add profiles to the menu "Vayping-> Modes" which is already in the circle! Then, using a switch Y/N, you can leave in a circle for example, three profiles and Bypass, or varivatt and two tanks profile on thermal control, or even how anyone!

And the name of the profile, I think is best seen in the uppermost row, and not at the bottom where the battery!

Thanks again @ClockSelect because he listens to our posts!

ClockSelect, 2016-11-14:
@GeorgeEdward : and what about Fire-(+) during two seconds?

ClockSelect, 2016-11-14:
@Villista : Fire-(+), select, click. Easy enough I think. It's not really worth writing a configuration menu, IMO.

GeorgeEdward, 2016-11-14:
@ClockSelect. I and many users don't like combinations with two buttons on mods like Fire-(+) cause you need two hands for pressing. It would be more convenient to press for example two times Fire button to enter the Profile selection screen and then choose one profile. In this case you need only one hand and one finger )))

GeorgeEdward, 2016-11-14:
Fire-(+) let's leave as another way to enter Profile selection screen.

Derducken, 2016-11-15:
@ClockSelect Whatever way they work, only for the fact that the profiles work, and we have them in our mods, I really-really want to thank you for their addition - and everything else you've done up to now. I really didn't expect you not only to implement them as fast as you did, but also to keep tweaking them to perfection.

tl;dr: Kudos. Many of them.

closed 2016-11-15 by ClockSelect