ClockSelect/myevic #148: nice firmware, any plan to remove rom gaming to feature to improve feature tx

zais1, 2016-11-09:
nice firmware, any plan to remove rom gaming to feature to improve feature tx

ClockSelect, 2016-11-09:
Things will be removed in a near future, indeed.
Game, SMART mode, glyphs...

For built-in batteries, I use Joyetech's curve since I don't know the true discharge curve of those batteries, and unless someone disassemble the box to benchmark the battery there's no way to know.

Unless you vape around 5W, I don't really see the point of small increments. Maybe replacing the 0.1 increment by 0.5 would make sense; the 0.1 is useless, imo. That's questionable.

SEEKNDESTROY, 2016-11-09:
I'm all for removing useless features to make room for what really matters.

zais1, 2016-11-09:
yes seems agreed take care

phantasm, 2016-11-09:
@ClockSelect 
if < 5W, use 0.5 step

ReMicroN, 2016-11-14:
Using <5W on 200W 3-battery boxmod?=)

LockManipulator, 2016-11-15:
I love having flappy bird, maybe make it optional to put on? I would hate to lose it.

ClockSelect, 2016-11-15:
@LockManipulator : A spare Nokia 1300 can be found for less than $10 nowadays.

LockManipulator, 2016-11-15:
It's more of a "cool factor" for me to have it on my vape. If the general consensus is it should be removed for other features though, I can't argue. Is there a way I'd be able to put it back on myself in lieu of the other features?

ClockSelect, 2016-11-15:
@LockManipulator : You can have your own fork of the project, if you wish, and keep/include whatever you want. But Flappy Bird is not gone yet.

zais1, 2016-11-15:
does less rom gives better power longevity?

ClockSelect, 2016-11-15:
@zais1 : no, it has no influence on power consumption.

ClockSelect, 2016-11-16:
We'll come back later on this. I remember the power thing.

closed 2016-11-16 by ClockSelect