ClockSelect/myevic #150: BVO Voltage Correction is not linear. Imbalance grows again while discharge.

ReMicroN, 2016-11-09:
Hello! I have RX200S and I charge batteries to 4.20 Volts with opus charger. To correct boxmod voltage I use this parameters:
+0.02
+0.01
+0.10
and boxmod shows me correct voltage 4.20 4.20 4.20
But when it discharged imbalance grows again. External charger shows same voltage on each battery.
With equally discharged to 3.00V batteries I need to use -0.05 +0.10 +0.02 radically different BVO setting to make boxmod show me same 3.00 voltages.
Batteries are same, from one batch. Samsung 25R.

ReMicroN, 2016-11-09:
Is it possible to add 2-stage BVO calibration.
With fully charged to 4.20 V
With discharged to 3.00 V (cut-off voltage)

ClockSelect, 2016-11-09:
Offset is not constant over the discharge, but I don't know if it's linear or not. I'd expect it's more or less linear, in fact.
Having an offet *and* a slope would be the right thing, but that's three more precious parameter bytes and a complicated UI to write and explain.
For now, BVO is a compromise; choose for wich voltage you prefer the batteries to be accurately displayed. If you choose accuracy for high voltages, the USB charging will correctly charge batteries up to their max capacity and won't stop the charge prematurely by reading a battery too high. If you choose low voltages, the box won't lock vaping prematurely by reading a battery too low.
If you use an external charger, choose low voltages, and you're fine.

ReMicroN, 2016-11-09:
So it will be great to add second "BVO USB" option.
When connect USB cable to boxmod it will change BVO settings "to accuracy for high voltages", when disconnect it will change to "low voltages" settings.

SIimJim, 2016-11-09:
ReMicroN
Did you test the voltages on a multimeter as the charger is unlikely to be that accurate and I bet they are not all exactly 4.20 volts :)

SIimJim, 2016-11-09:
For instance

[image]

ReMicroN, 2016-11-09:
Yes, I did. OPUS BT-C3100 shows me 4.20 Volts, multimeter - 4.17 Volts.

phantasm, 2016-11-09:
@ReMicroN @ClockSelect i agree, it should be 2 separate menus: BVO @full charge and let's say BVO @30% charge (or 20%, or anything low). you enter offsets for both modes, and device extrapolates the current value. readings can still differ from real value, but it will be much more accurate than with fixed BVO. but it's not an easy-breezy improvement, I think ;)

ClockSelect, 2016-11-12:
And imagine it's not linear?
We'd need a third menu. And if ever it's not quadratic? We're doomed!
No, seriously.
Things work no-so-badly by default.
Even with their imperfections, boxes allow you to exploit 80% of their capacity, and I think it's absolutly fine. There's almost nothing to gain in some millivolts here and there. The BVO is made to correct obvious skewed voltage readings, not to change your box in a precision multimeter. My rule of thumb is: if it blinks, change the battery.

closed 2016-11-12 by ReMicroN