ClockSelect/myevic #151: 12 hour clock format

damien9818, 2016-11-09:
awesome work on the firmware but is there any possibility adding a menu option to switch to a 12 hour digital clock format ?
example 8:44 PM and or 8:44:44 PM 
as well as the 24 hour format would be greatly appreciated 
thank you

ClockSelect, 2016-11-10:
Why not, but for consistency and readabilty, I must first allow vaped liquid to de displayed in fluid ounces and vaping power in foot-pounds per minute. I'll deal with the clock issue just after.
^^

ClockSelect, 2016-11-11:
I made date formats because dates may be confusing, between day and month. The problem does not exist with time; a 24h format is readable without ambiguity by the whole world. So I'll keep it as it is.

closed 2016-11-11 by ClockSelect

damien9818, 2016-11-11:
fair enough and thank you for giving an explanation

lordage, 2017-01-14:
so a 12 hour format is a no go.