ClockSelect/myevic #152: request please

FlorinDiablo, 2016-11-09:
Hi,

Could you please help me to remove intuitiv bare for counting vape second?
Tnx!

ClockSelect, 2016-11-09:
I'm afraid I don't understand the question... ?

Daveee10, 2016-11-09:
I think he mean the Pico style puff timer bar.. Maybe he want to switch off/change back to original style.. Correct me if I'm wrong... I don't understand why not good the new bar... IMO more eye friendly..

RenegadeTitan, 2016-11-09:
my guess is he cant find option for not displaying this bar

ClockSelect, 2016-11-10:
Puff timing is a legacy of old mods. Its usefullness is highly questionable. But if I remove it, I'll get two issues tomorrow asking how they can restore it. And I don't want to spend a config memory for that. I'll keep that for a later revamp of screen display; for now, I'll keep things as they are.

closed 2016-11-11 by ClockSelect