ClockSelect/myevic #153: Temperature board sensor

MicroSur, 2016-11-09:
Are You experimented with the possibility of using a board temperature sensor for the purpose of setup a temperature control? 
If you apply a correction constant in it, as is done for the batteries, the sensor can be used when user Lock resistance for TCR (not 20(C) everytime).
After prolonged shutdown off, sensor reading will match the ambient temperature.

afol2, 2016-11-09:
As i have read here, the Board Temperature its not accurate

falconettt, 2016-11-09:
I checked through the fridge and warm water. The board temperature is measured accurately enough. The misunderstanding arises from the fact that the board temperature and the surface temperature of the box are two different things.
/Vtwo mini, VTC mini/

phantasm, 2016-11-09:
board temp can (and will) differ from coil temp, especially if you use a factory coil head which has its own metal case and heat capacity, plus e-liquid temp, mod construction, RBA constrution, etc.

heat dissipation will be different every time, for every box mod. and if you use more that one RBA, it will be even more complicated.

personally, I don't think it's a good idea at all.

MicroSur, 2016-11-10:
Idea was not about coil temp while firing, but right after long suspend it will match to board temp.

ClockSelect, 2016-11-10:
I see what you mean.
It looks like a smart idea, but I'm not sure it's worth the effort. 
To calibrate the sensor, you'd need a thermometer. I you have one, you can simply adjust the resistance by a few milliohms to match the ambient temp.
Also, depending on your build, even using the same metal, your real TCR may vary significantly (theorical TCR are measured on calibrated straight wires that don't stritcly behave like coils).
If your goal is to achieve a 1°C accuracy on displayed temp, there's more work than reading the ambient temp.

dabugen, 2016-11-11:
Don´t forget the inertia of that sensor, since it is placed far away from the coil. Once you heat up the coil, it takes at least 3 to 5 seconds until the on-board sensor follows the raise in temperature, often much longer. Once it has reached a significant temperature, the coil will already be completely burned.

ClockSelect, 2016-11-11:
@dabugen : That's not the idea. The idea is to read the temp sensor at a moment when the box has been idle long enough so that it reads the ambient temp, and the coil is also at ambient temp, so the rez/temp couple would be calibrated.

ClockSelect, 2016-11-12:
I really think it'd cause more issues than it solves.
With the mΩ accuracy I did a few times ago, temp control achieves a real good performance. Being offset by ±5°C is not an issue, IMO.

dabugen, 2016-11-12:
Agreed!

closed 2016-11-13 by MicroSur