ClockSelect/myevic #154: Icons instead of Text

MoDuLah, 2016-11-10:
**Box Model:**
Evic VTC Mini
**Coil setup:**
Nife48, 28mm @0.63Ω

Back in the stock firmware i used to make some text swap and display icons instead of text for coil/power/puffs/ohms etc. 
[image]

But when i tried to put them on on "seemed" unused parts of fonts i got a scrambled result.
Is it still possible?

ClockSelect, 2016-11-10:
Sure it's still possible, it's just that some former unused glyphs are now used (0xD4 and 0xD5 in your picture).
You can easily fix this since the + and - signs now exist in 0xFC and 0xFD so move your 0xD4 to 0xCD, 0xD4 to 0xCF, change the strings accordingly, and you should be fine.

MoDuLah, 2016-11-10:
i'll give it another go and will let you know. :)

MoDuLah, 2016-11-10:
Done!!! Wish the transition to your next firmware to be so easy...
[image]

closed 2016-11-10 by MoDuLah