ClockSelect/myevic #156: Need lower setting for ml/kJ please (lower than 200)

dabugen, 2016-11-10:
Hi,

even though I´ve set ml/kJ to 200 already, it still shows to much total used liquid (not talking about ml/Day). Yesterday I did reset the "ml" counter and have now just used ~5ml in reality, still the ml counter says 8ml. But I can´t set it lower than "200", which I´d need to make that work for my scenario. So can you please make it setable to lower than 200 (maybe to 100 or something).

Thank you:)

ClockSelect, 2016-11-10:
Why not.

closed 2016-11-11 by ClockSelect

afol2, 2016-11-11:
How do i Calculate the correct value for me ? @ClockSelect 

Oumier Whitebone with Single parallel coil of Kanthal 25gauge, and Clapton of 26 gauge and 36 gauge, 4 wraps @  0.232 Ohms.
Power mode at 45watts and a preheat of 55watts at 0.3 segs
Evic VTC mini

ClockSelect, 2016-11-11:
You have to test.
Vape a known quantity of liquid, see the displayed result, and adjust the value proportionaly.

enricorov, 2016-11-11:
Do you think is feasible to implement a 'calibration' feature?

ClockSelect, 2016-11-11:
There's no way to do that.

afol2, 2016-11-11:
ok, I'll do that

enricorov, 2016-11-11:
@ClockSelect Why? Because of space constraints?

SEEKNDESTROY, 2016-11-11:
@enricorov There are too many different variables... Number of coils, types of metals, coil designs, wattage or temperature setting, inhale time etc.