ClockSelect/myevic #158: 3D screensavers

RenegadeTitan, 2016-11-11:
RX2/3 dodecahedron motion is jumpy now since last update. dont know if it is intentional but i think its an improvement. just pointing it out in case its a symptom of a different issue. 

one idea i had for screensaver is 4D cube where time represents part of 4th dimension. maybe with some rotation. crude examples can be found on youtube under "4D cube" search. i believe examples can be improved upon.

ClockSelect, 2016-11-12:
It was the first anim I did on an Amiga 4000, long ago.
N-dimension animations, all matrix multiplication holding in the internal 68030 cache.
Souvenir, souvenir...
You'll forgive me if I decide there's not enough buttons on the box the control angles?

RenegadeTitan, 2016-11-12:
lol thats awesome.

yeah man, just something i thought was cool. seemed like it could be a good addition so brought it to your attention. im not really attached to the idea.

ClockSelect, 2016-11-12:
I'd like 4D as much as you but it takes space and cpu cycles (n²).
A tesseract would take for itself the whole five platonic solids or even more.
I think things are fine, for now. Let people the time to be bored.

closed 2016-11-12 by ClockSelect