ClockSelect/myevic #160: Suggestion: double click as "back" in menus?

Derducken, 2016-11-12:
I don't know if it's possible to detect double clicks while in menus, like when on the "main screen", but I think it would be easier (even a tiny bit) having a double click as a "back" shortcut instead of having to press power and right at the same time.

If not, could we have a "Back" instead of "Exit" at the end of each menu, lead to the *previous menu* instead of "quitting tweaking"?

My reasoning; let's say you want to tweak "the look of the screen". You've got to enter-exit-enter-exit three-four submenus (brightness, logo, screensaver). Each time you either re-enter the same sub-menus, or use a two-button shortcut, when you'd just have to click "back" right after the last option you edited.

Of course, that's the way *I* find it logical. You know what they say about opinions :-)

ClockSelect, 2016-11-12:
Both buttons is a back button.

Derducken, 2016-11-12:
I know - I said "...or use a two-button shortcut" :-)

ClockSelect, 2016-11-12:
I Shall remove the "Exit" alltogether since Fire-+ is an exit.
So I'm more prone to remove things than to add things.
But I know if I remove the Exit people will panic, so I don't.
And if I add another Exit, people will hesitate and let them kept in fire. So I won't either.

Derducken, 2016-11-12:
Damn, you're right on both fronts.

It's just that I've worked years writing reviews and tutorials for software, and then as a site manager, and then as a web designer, and I'm thus used to the "hierarchical" model of menus, where when you enter the submenu of a submenu you always have access to the previous menu, not to exit the program or browser altogether :-)

I know, different things. As I said, it's just my opinion of "how it should be". Consider it wrong. And worthless :-P

RenegadeTitan, 2016-11-12:
i vote for reversing the exit and back. left right combo to exit. back option at bottom. also just my opinion.

ClockSelect, 2016-11-12:
The "back" in the menu system has been a debate since the very begining. You can't use the double-click since the click is instantly taken into account by a menu option if you want something responsive. And I didn't want a "Back" option in every menu either, because screen space is limited.
The double (+)-(-) to come back to parent menu was an Idea of the russians (thx maelstrom2001) I adopted since anyway never any thing you'd have to tweak would need both be more and less at the same time. I found it was smart, and now many people are used to it, so let's keep it.

drminol, 2016-11-12:
I think the better way is change "exit" in menu to "back". If you want exit when you are in submenus you can click fire and "+". What do you think?

enricorov, 2016-11-12:
I also think the same way. 'back' in submenus makes more sense.

phantasm, 2016-11-12:
I'm also voting for Tr... sorry, for "back" behavior instead of "exit".

ClockSelect, 2016-11-16:
When you need to tweak something, you enter the menu, tweak it, and exit, and vape. You don't "back", unless you're playing with menus. That's the great debate.

GD386, 2016-11-18:
I think VT users can't do a "+-"  because of the single rotary +- button

ClockSelect, 2016-11-21:
@GD386 good point.

pdf, 2016-12-30:
> When you need to tweak something, you enter the menu, tweak it, and exit, and vape.

Only if you know exactly which menu the thing you're looking for is (as a developer, you probably have them all memorized, as a user... not so much), otherwise you enter the menu, enter the wrong submenu, exit, enter the menu again, try another menu, find the wrong one, repeat until you find what you're looking for.

Definitely would prefer back to exit.

ClockSelect, 2017-01-18:
Let's give it a try.

SEEKNDESTROY, 2017-01-18:
@ClockSelect It still says exit in contrast setting. 
I'm not a huge fan of the "Back" feature, too much button pressing to exit the menu... can you set it so pressing the UP & Down buttons exits the menu?

Thanks for everything you do.

ClockSelect, 2017-01-18:
Fire-<+> enters *and exits* menus. Fast and handy.

<->-<+> combo still go back, but this combo should be avoided due to some boxes having such a button that you cannot press both at the same time (like the VTwo).

closed 2017-01-21 by ClockSelect